espressif/esp-rainmaker #184: ESP 32 BLE and SoftAP connection problem Device not found

zakirox123, 2022-10-10:
i am using ESP-WROOM-32 Device with Rainmaker App and having a problem tried with BLE and SoftAp, no luck BLE Error message "Communication Failed" sometime it says "Cannot find the device"
SoftAP Error message "Assisted Claiming not supported for SoftAP. Cannot Proceed" again tried to flashed with new firmware using espressif firmware tool still no luck. Tried to connect with nRF Connect App, got an error in log "Error (0x85): GATT ERROR" if i use auto connect it will get connect after long time to nRF Connect App

`#include "RMaker.h"
#include "WiFi.h"
#include "WiFiProv.h"

const char *service_name = "PROV_12345";
const char *pop = "123456";

// define the Device Names
char deviceName_1[] = "Switch1";
char deviceName_2[] = "Switch2";
char deviceName_3[] = "Switch3";
char deviceName_4[] = "Switch4";

// define the GPIO connected with Relays and switches
static uint8_t RelayPin1 = 23;  //D23
static uint8_t RelayPin2 = 22;  //D22
static uint8_t RelayPin3 = 21;  //D21
static uint8_t RelayPin4 = 19;  //D19

static uint8_t SwitchPin1 = 13;  //D13
static uint8_t SwitchPin2 = 12;  //D12
static uint8_t SwitchPin3 = 14;  //D14
static uint8_t SwitchPin4 = 27;  //D27

static uint8_t wifiLed    = 2;   //D2
static uint8_t gpio_reset = 0;

/* Variable for reading pin status*/
// Relay State
bool toggleState_1 = LOW; //Define integer to remember the toggle state for relay 1
bool toggleState_2 = LOW; //Define integer to remember the toggle state for relay 2
bool toggleState_3 = LOW; //Define integer to remember the toggle state for relay 3
bool toggleState_4 = LOW; //Define integer to remember the toggle state for relay 4

// Switch State
bool SwitchState_1 = LOW;
bool SwitchState_2 = LOW;
bool SwitchState_3 = LOW;
bool SwitchState_4 = LOW;

//The framework provides some standard device types like switch, lightbulb, fan, temperature sensor.
static Switch my_switch1(deviceName_1, &RelayPin1);
static Switch my_switch2(deviceName_2, &RelayPin2);
static Switch my_switch3(deviceName_3, &RelayPin3);
static Switch my_switch4(deviceName_4, &RelayPin4);

void sysProvEvent(arduino_event_t *sys_event)
{

    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32

 Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
       printQR(service_name, pop, "ble");
       
        // digitalWrite(wifiLed, true);
#else
       Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
         
#endif        
        break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
        Serial.printf("\nConnected to Wi-Fi!\n");
        digitalWrite(wifiLed, true);
        break;
[... 207 lines cut ...]
          RMakerWiFiReset(2);
        }
    }
    delay(100);

    if (WiFi.status() != WL_CONNECTED)
    {
      //Serial.println("WiFi Not Connected");
     // digitalWrite(wifiLed, false);
      ledBlink();
    }
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
    }

    manual_control();
}

void ledBlink(){
    digitalWrite(wifiLed, true);
    delay(300);
    digitalWrite(wifiLed, false);
}`

sanketwadekar, 2022-10-11:
HI @zakirox123,
I did not get any errors while provisioning the device over BLE. I have tried this on ESP32-Arduino v2.0.5 and the Rainmaker Android Phone app. Which version of ESP32 Arduino are you using?

zakirox123, 2022-10-13:
> 

I tried with the latest ESP32-Arduino v2.0.5 and ESP32 DEVkit V1

Tonvina, 2022-11-18:
don't know if you solved this problem i had this problem too but on esp32s

zakirox123, 2022-11-21:
> 

no still no luck i am finding some alternative solutions

khushbushah2302, 2022-11-21:
Hi @zakirox123 , 
Can you try once after disabling [CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION](https://github.com/espressif/esp-idf/blob/01d014c42def8d0c19e1ce55c07de6761e092ffa/components/wifi_provisioning/Kconfig#L34), which you can find under Component Config -> Wi-Fi Provisioning Manager ?

pablovm43, 2024-04-19:
Hola Pudieron solucionar este problema?, tengo el mismo problema.