espressif/esp-rainmaker #186: errors through provisioning

aymancom, 2022-10-16:
I got these errors when I connect the esp with internet form rainmaker through the provisioning
this error occurs when it makes confirming wi-fi connection:
E (31318) esp-tls-mbedtls: mbedtls_ssl_setup returned -0x7F00
E (31319) esp-tls: create_ssl_handle failed
E (31319) esp-tls: Failed to open new connection
E (31349) TRANSPORT_BASE: Failed to open a new connection
E (31412) MQTT_CLIENT: Error transport connect
E (31461) esp_mqtt_glue: MQTT_EVENT_ERROR

and this error occurs when it makes confirming node association:
E (56615) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
E (56615) esp_mqtt_glue: MQTT_EVENT_ERROR
E (56615) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (56690) MQTT_CLIENT: MQTT connect failed

any one can help?

shahpiyushv, 2022-10-17:
@aymancom , can you tell what esp idf version you are using and also check if you see any Wi-Fi connection errors before/after this? Please also let us know the chip you are using.

aymancom, 2022-10-19:
I`m using esp-wroom-32, and no I`m not seeing any Wi-Fi errors.
This problem happens for only one chip.