espressif/esp-rainmaker #189: Error: Confirming Node Association Failed

UserOfSpace, 2022-10-18:
How do I fix this problem with my esp32 Dev Kit V1 board while trying it to connect to my phone?
(Note: I used Arduino IDE For the Coding & Uploading part & also I am a complete beginner so please do help me in the way I could understand.)
This Is the code: https://github.com/techiesms/All-in-One-Home-Automation–Version-2-

shahpiyushv, 2022-10-21:
@UserOfSpace , do you see your device crashing and rebooting during provisioning? If yes, please `#include <wifi_provisioning/manager.h>` in your code and add this into the [event handler](https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino#L28)

```
  switch (sys_event->event_id) {
    case ARDUINO_EVENT_PROV_INIT:
      wifi_prov_mgr_disable_auto_stop(10000);
      break;
    case ARDUINO_EVENT_PROV_CRED_SUCCESS:
      Serial.println("Stopping Provisioning!!!");
      wifi_prov_mgr_stop_provisioning();
      break;
```

UserOfSpace, 2022-10-29:
> @UserOfSpace , do you see your device crashing and rebooting during provisioning? If yes, please `#include <wifi_provisioning/manager.h>` in your code and add this into the [event [...]

.

closed 2022-11-13 by UserOfSpace

anjalir14, 2022-12-23:
I'm having the same issue but my device is not crashing and rebooting it's just not able to connect with mqtt.