espressif/esp-rainmaker #190: Connectivity status

b4sudhir, 2022-10-21:
How do I know or differentiate if my IOT device is connected to wifi without internet. 
Use cases : Connected to wifi, with internet, Connected to wifi without internet(when WAN is failing).

shahpiyushv, 2022-10-21:
From RainMaker's perspective, connected to Internet = Connected to MQTT.

So, for checking Internet connectivity, you can monitor the `RMAKER_MQTT_EVENT_CONNECTED` and `RMAKER_MQTT_EVENT_DISCONNECTED` events as shown [here](https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L84)

b4sudhir, 2022-10-21:
I am unable to access event handler objects or class members. Can you please guide me with some example. i am trying to access ESP_EVENT_DECLARE_BASE for RMAKER_MQTT_EVENT_CONNECTED this enum. any example from this will help which i was not able to find anywhere .
Thank you!!!

shahpiyushv, 2022-10-21:
The example pointer is already shared above. Sharing again

Registering for events: https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L139

Handling events: https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L84

You need to include these files

```c
#include <esp_event.h>
#include <esp_rmaker_common_events.h>
```

b4sudhir, 2022-10-26:
Hi there !
Thanks for the  help...resolved above problem by your given example. 
Stuck in next issue which is :

    Esp Rain maker doesn't reconnect after internet is down. Wifi is still connected  only internet is down for some time and after internet came it doesn't connect to the mqtt server and also feel like esp rainmaker is unstable as my esp32 is rebooting many times randomly, not frequently but like twice in a day. 

My Assumption are : 
    Rainmaker is not able to reconnect mqtt unless wifi disconnect event is triggered.
    Rebooting may be occurring due to added IR code with esp rainmaker and might be not compatible with esp rainmaker.
    
Any suggestion for reconnect code will be appreciated.  Thanks in advance.

shahpiyushv, 2022-10-26:
@b4sudhir , MQTT client keeps retrying after every 10 seconds, irrespective of what the disconnect reason was. It could be Wi-Fi disconnect or Internet lost or server itself being down. One reason for the connection failing could also be low memory.

You can use the below snippet and call it from the MQTT disconnected event to see how much RAM is available

```
#include <esp_heap_caps.h>
static void print_heap(void)
{
    Serial.printf("\tDescription\tInternal\tSPIRAM\n");
    Serial.printf("Current Free Memory\t%d\t\t%d\n",
           heap_caps_get_free_size(MALLOC_CAP_8BIT) - heap_caps_get_free_size(MALLOC_CAP_SPIRAM),
           heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
    Serial.printf("Largest Free Block\t%d\t\t%d\n",
           heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
    Serial.printf("Min. Ever Free Size\t%d\t\t%d\n",
           heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_minimum_free_size(MALLOC_CAP_SPIRAM));
}
```

Can you tell if you are using Arduino or esp idf for RainMaker? If you are using esp-idf, just replace `Serial.printf(` with `printf(` or `ESP_LOGI(TAG, `

b4sudhir, 2022-10-31:
@shahpiyushv Thank you So much.
New Query : How to get changed device name edited by app using edit Name. I want to save the edited name EEPROM so that on next reboot device name will remain same. Also can i get the mqtt message send or received by device(Getting mqtt message id by above technique).

shahpiyushv, 2022-10-31:
@b4sudhir , generally, it is recommended to add a new issue if it is not related to the current one. That helps in better tracking and can also help other users who are looking for similar answers :)

The name is a normal parameter like rest of the parameters, with the only difference that it is handled internally. However, you can enable [CONFIG_ RMAKER_NAME_PARAM_CB](https://github.com/espressif/esp-rainmaker/blob/master/components/esp_rainmaker/Kconfig.projbuild#L82) so that you get callback for that too and handle it yourself.

For MQTT publish, you can use [RMAKER_MQTT_EVENT_PUBLISHED](https://github.com/espressif/esp-rainmaker/blob/master/examples/switch/main/app_main.c#L90) event. We have not exposed the message received events via RainMaker because those are redirected to appropriate callbacks. Meanwhile, just curious to know what your use case is.

shahpiyushv, 2022-11-03:
@b4sudhir , even right now, is you use the default name parameter (which is automatically added in the standard devices) the name is stored in persistent storage (nvs) on the device side and so, is retained across a reboot. Is there any specific use case for storing it in EEPROM instead?