espressif/esp-rainmaker #191: esphome (or Home Assistant) integration

trombik, 2022-10-31:
I've just started exploring ESP RAINMAKER. I do not have clear requirements, or how the cloud system, esp nodes, and Home Assistant (HA) should work. still, I'm thinking about the possibilities.

I would like to ask how the developers, and the management, see esphome and HA. are they an enemy, or a possible ally? while epshome is focusing on the esp32 family, HA is not. HA, and its vendor, is kind-of competitor. while HA is trying to integrate as many third-party products as possible, espressif is trying to sell more esp chips. "3rd Party Integrations" in the official doc only refers to GVA and Alexa. there are other home automation solutions, in which HA is quite popular. what do you think about this ind of "3rd Party Integrations"?

As an esphome user, what I like in rainmaker is:

* the visualization of metrics (HA is not good at this, and I am using influxdb and grafana for that)
* device and log management (crash on a hard-to-access switch is annoying and hard to debug)

my list of would-be-nice-to-have is:

* integrate rainmaker component into esphome without losing any functionalities of both (so that I can control devices as esphome node and rainmaker node)
* control switches, or lights, from both HA and rainmaker phone app

pros for espressif:

* many of esphome users are a heavy user; they implements esphome components or HA integrations if they don't exits (me included, i'm a maintainer of esp-idf components and esphome components)
* their home system can be a good testing field (I have /24 network for nodes, but i'm thinking about expanding the subnet)

cons for espressif:

* esphome is designed for a standalone system; a common belief among esphome users is that, IoT devices should work without access to the Internet (some are against cloud solutions)
* HA has already implemented many components (music players, light controls, etc), which cannot be found in rainmaker (and the interface, or APIs, has to match the one in HA)

kedars, 2022-11-02:
> I would like to ask how the developers, and the management, see esphome and HA. are they an enemy, or a possible ally?

Ally, for sure :)

We understand that different users care about specific things, the more choice there is, the better for everyone. And we're happy to assist as much as possible. 

> device and log management (crash on a hard-to-access switch is annoying and hard to debug)

Note that this is a separate service from ESP RainMaker called ESP Insights  (https://insights.espressif.com), and it can be used in conjunction with any other framework quite easily. 

ESP RainMaker primarily offers 2 things (a) controlling devices remotely as end-users (b) device-mgmt (OTA etc) as device-makers. From my understanding of the HA ecosystem, it offers these options to the users in their local network, so I am not sure adding HA support into RainMaker helps in this context. 

One of the things we could possibly do is do the other way round: write a HA plugin for RainMaker so that HA can control RainMaker based devices with ease. Would that help? 
Do let us know if there are specific feature requests that you think will help, and we'd certainly love to understand how we can extend RainMaker to better fit the requirement.

trombik, 2022-11-02:
> Ally, for sure :)

Glad to hear that.

> Note that this is a separate service from ESP RainMaker called ESP Insights (https://insights.espressif.com), and it can be used in conjunction with any other framework quite easily.

Got it.

> ESP RainMaker primarily offers 2 things (a) controlling devices remotely as end-users (b) device-mgmt (OTA etc) as device-makers. From my understanding of the HA ecosystem, it offers these options [...]

HA does automatically discover esphome devices, and create all the buttons, sliders, or any UI widgets in the UI. However, it does not manage epshome devices, logs, or OTA. It is a responsibility of esphome's web interface, but the interface does not scale very well. hence, I need a solution.

> One of the things we could possibly do is do the other way round: write a HA plugin for RainMaker so that HA can control RainMaker based devices with ease. Would that help?

It definitely would! HA has MQTT integration, and tasmota is a longtime user of it.

> Do let us know if there are specific feature requests that you think will help, and we'd certainly love to understand how we can extend RainMaker to better fit the requirement.

Yes, I would. I'm trying to understand RainMaker, and its possibilities. When I finish, I will create a list of "would-be-nice" features and missing bits in RainMaker to implement them. Personally, I would not keep all the eggs in a cloud, which is why I have a local HA server, but not all the people is geeky enough. I also would like to help espressif because the company helped me a lot in my development (writing good docs, providing disciplined code, giving quick responses to questions, etc) .

closed 2022-11-02 by trombik

JohnMacrae, 2022-11-02:
Home Assistant can already control Rainmaker devices through a local MQTT
bridge. See https://youtu.be/Ta9dwDD9GyA

inuonly, 2023-05-18:
> Home Assistant can already control Rainmaker devices through a local MQTT bridge. See https://youtu.be/Ta9dwDD9GyA
> […](#)

As a hack this might work, but you need more resources here esp nodes to be wasted, and since esp rainmaker is already on the matter train, it might be easy to ask the team to built mqtt support/integration so that it be any automation , like home assistant, or openhabian, or others it will be a simple mqtt calls to esp rainmaker node which is already widely used. Point being baking such (since esp rainmaker is rapidly growing and improving so life cycle is very much active for development) we can ask this to be a native feature. Although your link is still very helpful , so thanks for that as it still solves the issue for time being while the team is trying to bake in such features, It will be not practical without these home automation system to use everything at one place imagine changing evertything in respective applications of each node and device you have at home ;) All these arguments I made are hopefully worthy to consider adding mqtt or similar plugin as suggested above to integrate it to automation systems, That will be my one vote for the same.