espressif/esp-rainmaker #195: Delete nodes from dashboard and getting json data in my notifications

Jo14532, 2022-11-15:
I would like to remove a few from my dashboard, as I will not be using them anymore. How do I do this?

[image]

Secondly, I set up my ESP32 to send alerts, This worked for the first few weeks flawlessly, however now I hardly get them. If I get them they also appear in a JSON format on my phone as below. Why is this and why do I not receive them every time I trigger it?

[image]

shahpiyushv, 2022-11-15:
@Jo14532 , removing nodes from dashboard is not possible right now because of some complexity in that and also due to some legacy reasons. However, we are working on adding support for that in the backend. As of now, If you have reached your quota, you can get it increased by sending an email to esp-rainmaker-admin@espressif.com from your registered email address.

We are working on fixing the other issue about the notification showing up as JSON and it will be rolled out asap.

Jo14532, 2022-11-15:
> @Jo14532 , removing nodes from dashboard is not possible right now because of some complexity in that and also due to some legacy reasons. However, we are working on adding support for that in the [...]

What is the quota, I didn't see any mention of this in docs

shahpiyushv, 2022-11-15:
@Jo14532 , the concept of quota is meant to ensure that the public RainMaker is used only for hobby, educational and evaluation purposes and not for larger fleets. Moreover, the quota limit applies only to assisted and host claiming. If you are using esp32c3, esp32s2 or esp32s3 with self claiming, the quota limit won't apply because these chips have a unique id which gets validated during claiming to ensure that it is indeed an ESP device.

Jo14532, 2022-11-15:
> @Jo14532 , the concept of quota is meant to ensure that the public RainMaker is used only for hobby, educational and evaluation purposes and not for larger fleets. Moreover, the quota limit applies [...]

Thanks, I'm using the esp32 wroom chip, what will be my quota for this chip?

shahpiyushv, 2022-11-15:
@Jo14532 , the quota for esp32 is 5 and as mentioned in [my earlier comment](https://github.com/espressif/esp-rainmaker/issues/195#issuecomment-1315113281), you can get it increased.

Jo14532, 2022-11-25:
> @Jo14532 , removing nodes from dashboard is not possible right now because of some complexity in that and also due to some legacy reasons. However, we are working on adding support for that in the [...]

Is there any timeline on when the latter will be resolved?

shahpiyushv, 2022-11-28:
@Jo14532 , the notification issue is now fixed. Please check.

closed 2022-11-28 by Jo14532

Jo14532, 2022-11-28:
> @Jo14532 , the notification issue is now fixed. Please check.

Thank you. I am now receiving the correct formatted messages!

shahpiyushv, 2022-11-29:
@nevi-me ,

> Hi @shahpiyushv , this might be related to the original question in the issue. I seem to have messed up my device while provisioning it. It shows as a node, but has an "invalid chip"

This "invalid chip" info is the ESP Insights dashboard and it will show up only if you want to use Insights and have enabled it using `CONFIG_ESP_INSIGHTS_ENABLED=y`

Marcarleto, 2023-01-31:
> @Jo14532 , the concept of quota is meant to ensure that the public RainMaker is used only for hobby, educational and evaluation purposes and not for larger fleets. Moreover, the quota limit applies [...]

@shahpiyushv at my company here we were planning to use it for larger fleets, so isn't it possible? Is there any kind of paid plan to support us?

sanketwadekar, 2023-01-31:
Hello @Marcarleto 
We have paid plans for private deployment.
For commercial enquiries, you can reach out to the Espressif sales teams via the "Contact Us" option on https://rainmaker.espressif.com/ or by sending an email to [sales@espressif.com](mailto:sales@espressif.com).

Reubenruby7777, 2023-05-04:
> @Jo14532 , removing nodes from dashboard is not possible right now because of some complexity in that and also due to some legacy reasons. However, we are working on adding support for that in the backend. As of now, If you have reached your quota, you can get it increased by sending an email to [esp-rainmaker-admin@espressif.com](mailto:esp-rainmaker-admin@espressif.com) from your registered email address.
> 
> We are working on fixing the other issue about the notification showing up as JSON and it will be rolled out asap.

There's a chance to remove all the nodes permanently by deleting the account and signing in again.

shahpiyushv, 2023-05-05:
@Reubenruby7777, there is also an option to delete the nodes directly from the dashboard now.

Reubenruby7777, 2023-05-05:
> @Reubenruby7777, there is also an option to delete the nodes directly from the dashboard now.

You're right! But that won't be able to clear the quota by removing the nodes from the dashboard.

shahpiyushv, 2023-05-06:
> You're right! But that won't be able to clear the quota by removing the nodes from the dashboard.

@Reubenruby7777 , deleting from dashboard indeed removes the node from your claimed node list and helps regain the lost quota. If that isn't happening for you, do let us know.

Reubenruby7777, 2023-05-06:
That's not happened in my case, I've deleted the nodes from esp rainmaker and I have only one active node, but when I try to add another node it showed the quota exceeded and contact the administrator

Jo14532, 2023-05-06:
How do I check how many devices are left as per the quota?
--
Kind regards,

S.J Naidoo (MSc Eng(Electronic), UKZN)
Contact number: 078 7125 195

Reubenruby7777, 2023-05-06:
> > You're right! But that won't be able to clear the quota by removing the nodes from the dashboard.
> 
> @Reubenruby7777 , deleting from dashboard indeed removes the node from your claimed node list and helps regain the lost quota. If that isn't happening for you, do let us know.

That's not happened in my case, I've deleted the nodes from esp rainmaker and I have only one active node, but when I try to add another node it showed the quota exceeded and contact the administrator, I've tried to delete my account and signed up again, I got my quota back and everything works fine now.