espressif/esp-rainmaker #197: Cannot find addon for ESPRainmaker to integrate it to HomeAssistant

inuonly, 2022-11-20:
I searched for addon for ESP Rainmaker integration to HomeAssistant but could not find it. If someone already knows any please share the link or else if there exists none then please consider this issue of not being able to integrate with HomeAssistant.

JohnMacrae, 2022-11-20:
You can use Rainmaker to control any device that you have in Home Assistant
that uses MQTT with a simple MQTT bridge device. See here for details:
https://youtu.be/Ta9dwDD9GyA

Kind Regards
John

inuonly, 2022-11-20:
Hi John, 
This is exactly opposite of what I am looking, I am looking to integrate esp rainmaker into homeassistant. I do appreciate your efforts for a quick reply, I just need to reverse the role. I even checked custom controls but currently there is no documentation on how to integrate esp rainmaker to homeassistant. In simple words the existing controls (devices) vailable in rainmaker app should be able to be added(integrated) with homeassistant dashboard. Hope this simplifies any confision if left in original post. 
Thanks in advance.

inuonly, 2022-11-20:
Looks interesting, will try.

K2Here, 2024-02-24:
> hey @inuonly did you find any solution?
TLDR : No
nope, i even tried to host my own docker matter server for a workaround as a result of some online suggestions over internet after searchign on google, but to me nothing works as of now for me. After waiting for some time for any official support or announcemnt I gave up as there was no further support or development on this from their (rainmaker team) side. So not using rainmaker anymore .

sparsh308, 2024-09-24:
https://sparsh308.medium.com/home-automation-with-esp32-esp-rainmaker-and-home-assistant-with-manual-control-with-without-0a146fd0b189

My article it's little different but does same thing

sparsh308, 2024-09-24:
> hey @inuonly did you find any solution?
> TLDR : No
> nope, i even tried to host my own docker matter server for a workaround as a result of some online suggestions over internet after searchign on [...]

https://sparsh308.medium.com/home-automation-with-esp32-esp-rainmaker-and-home-assistant-with-manual-control-with-without-0a146fd0b189

shahpiyushv, 2024-09-25:
@inuonly , Home Assistant integration was not in our plans, but as pointed out by some users, the specs and APIs are open for people to play around with and add own integrations.

RainMaker is based on AWS serverless architecture and so, there is no concept of docker images as such. Private deployments are given out for commercial purposes based on discussions with our sales team. Alexa/GVA integrations and flashing tools (like admin CLI, claiming, pre provisioned modules) have been around for close to 4 years. So apart from Home Assistant integration, can you elaborate on what you were looking for which we could not provide?