espressif/esp-rainmaker #201: Apple M1 incompatible architecture (have 'arm64', need 'x86_64')

kiralikbeyin, 2022-11-28:
I can't compile and I'm stuck here.

````

[2/549] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle /Users/myMac/Documents/esp-rainmaker/examples/switch/build/esp-idf/mbedtls/x509_crt_bundle 
cd /Users/myMac/Documents/esp-rainmaker/examples/switch/build/esp-idf/mbedtls && /Users/myMac/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --filter /Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/cmn_crt_authorities.csv --input /Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q
Traceback (most recent call last):
  File "/Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 224, in <module>
    main()
  File "/Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 206, in main
    bundle.add_with_filter(path, args.filter)
  File "/Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 178, in add_with_filter
    self.add_from_pem(filtered_crts)
  File "/Users/myMac/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py", line 116, in add_from_pem
    self.certificates.append(x509.load_pem_x509_certificate(crt.encode(), default_backend()))
  File "/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/cryptography/hazmat/backends/__init__.py", line 14, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/__init__.py", line 6, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/cryptography/hazmat/backends/openssl/backend.py", line 113, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/cryptography/hazmat/bindings/openssl/binding.py", line 14, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: dlopen(/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so, 0x0002): tried: '/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (no such file), '/Users/myMac/.espressif/python_env/idf4.4_py3.9_env/lib/python3.9/site-packages/_cffi_backend.cpython-39-darwin.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
````

shahpiyushv, 2022-12-21:
@kiralikbeyin I just tried the compilation on a new M1 MacBook Pro, but I did not see any such errors. Can you specify the exact steps you followed?

yyeargan, 2022-12-28:
I just installed esp-idf onto a new MacBook Pro with an M1 Pro and experienced the same issue.

It seems the esp-idf install.sh script creates a Python virtual environment. The version of Python installed into that will be the same as the version of Python installed on MacOS.

When installing esp-idf using the default Python 3.8, I experienced the reported issue.

I then removed esp-idf, then installed Python 3.11.1 using macports and configured that version as the default for "python3", then installed esp-idf again.

Installing esp-idf under Python 3.11.1 worked as expected. I was able to do a fullclean and build without issue.

hobieboy, 2023-08-21:
Well, after deleting my build folder to do a clean build, I am running into exact same problem as the OP. I was using v4.4.3 then also tried installing a brand new v5.0.3, but both exhibit same problem. I have also tried updating my Python to v3.11.3 and v3.11.4 from the previous 3.9.3 but that didn't work either.

Anyone has found a solution for this? Now I am totally stucked ☹️

P.S. MacOS running is 13.3.1 on a M1 MacBook Pro