espressif/esp-rainmaker #202: Shared member does not receive notifications

Jo14532, 2022-11-28:
I added another user to access my ESP32 using the shared member feature on the ESP rainmaker android app.

When I press the switch button, I get a notification pop-up that says button pressed and the LED on the ESP32 board lights up. However, when the other user presses the switch button on his phone the LED does also light up but he does not get the pop-up notification but I do get it on my phone.

How do I grant the second user access to get these notifications?

shahpiyushv, 2022-11-29:
@Jo14532 , are you asking about the node alerts explicitly triggered from the firmware using `esp_rmaker_param_update_and_notify()` or `esp_rmaker_raise_alert()`? If yes, I just checked at my end and both, the primary and secondary users were getting those notifications. For normal param updates, no popup is expected anyways, since these are silent notifications, meant just to update the UI in real time.

Jo14532, 2022-11-29:
@shahpiyushv I am using the following:

[image]
 
On my phone when I arm the system I receive the following notification:

[image]

However, on the second users' phone, he does not but I do.

Jo14532, 2022-11-29:
@shahpiyushv hes using an Huawei P50 lite. So not running Android or ios. It's running Huawei OS

And no he does not get the online andd offline notifications as well

shahpiyushv, 2022-11-29:
We support only Google FCM (Android) and APNS (iOS) for push notifications. Seems the issue is seen because Huawei OS has its own push notification service. So, this does not seem to be an issue of secondary user, but of missing support.

Jo14532, 2022-11-29:
@shahpiyushv noted. So I guess the next question is will you guys add support for the Huawei OS or should I inform the secondary user to swop his phone if he wants the push notifications... (LOL)