espressif/esp-rainmaker #203: RMaker Agent connection to Rmaker Cloud

atanudasgupta, 2022-12-03:
At times the RMaker agent on the device loses connection to the cloud and does not reconnect, it could be due to the service provider rotates the dynamic IP addresses.

How do I check from the esp32 device that the connection is still on with RMaker cloud?

If I can detect the disconnection I can do RMakerWifiReset() and resolve this.

Using Arduino IDE with RMaker

sanketwadekar, 2022-12-06:
Hi @atanudasgupta,
Does [this comment](https://github.com/espressif/esp-rainmaker/issues/190#issuecomment-1286738548) solve your issue ?

shahpiyushv, 2022-12-06:
@atanudasgupta , meanwhile, is the issue really about MQTT disconnect or Wi-Fi disconnect?

atanudasgupta, 2022-12-06:
HI Piyush,
It is about Wifi disconnect, when the broadband provider changes the dynamic IP allocations. I see disconnect from the device to the cloud.
regards,Atanu

Atanu Dasgupta, Phone : +91 9901756756
Skype ID : ***@***.***

atanudasgupta, 2022-12-07:
Hi Sanket,I am using Ardruino IDE, will these header files given the example work?

regards,Atanu

Atanu Dasgupta, Phone : +91 9901756756
Skype ID : ***@***.***