espressif/esp-rainmaker #205: gpio_button

espressif2022, 2022-12-13:
**gpio_button** is necessary, can it be removed？
Now component [espressif/button](https://components.espressif.com/components/espressif/button) is used, there is a conflict；

shahpiyushv, 2022-12-13:
Older esp idfs do not use component manager by default and so, this component has been retained. It was renamed from button to gpio_button to prevent some conflicts in cmake dependency resolution. Can you elaborate what is the exact issue you are seeing? I believe your compilation must be finding 2 instances of the same APIs. Is that so?

espressif2022, 2022-12-13:
yes，compilation find 2 instances of the same APIs. 
I noticed that the function name has been changed here to prevent conflicts, but the newly added component espressif/button use the same name as you.

I'll ask the maintainer of button component if they can handle it.
Thanks!