espressif/esp-rainmaker #206: ESP32 does not connect to wifi extender

Jo14532, 2022-12-17:
I am having trouble in terms of my Wifi signal strength in my room. Hence I installed a Wifi range extender. However, I see that my ESP32 chip still does not connect to the Wifi. My extender and router share the same SSID and password.

Can someone assist why my ESP32 only connects to my main router but not to my extender?

shahpiyushv, 2022-12-18:
@Jo14532 , in esp-idf's wifi provisioning component, please try adding the below line after https://github.com/espressif/esp-idf/blob/master/components/wifi_provisioning/src/handlers.c#L113:

```
wifi_cfg->sta.scan_method = WIFI_ALL_CHANNEL_SCAN;
```

Let us know if you see any improvements with this.

Jo14532, 2022-12-18:
@shahpiyushv yes I can definitely this worked, see below results.

[image]

Before firmware change: -82dbm
After firmware change: -48dBm

shahpiyushv, 2023-01-13:
Closing this issue since it has been addressed. This change is available in esp-idf now (https://github.com/espressif/esp-idf/commit/bd6a74fcacc784b9abe80923b037debee73316d5)