espressif/esp-rainmaker #208: Failed at "confirming node association" (IDFGH-8955)

FlynnMa, 2022-12-19:


sanketwadekar, 2022-12-19:
HI @FlynnMa,
Can you include the device side "Error logs" that you get when "user node association" fails.
I've tried the switch example on idf v4.4 tag (8153bfe4125e6a608abccf1561fd10285016c90a) and did not see any errors while provisioning. Have you tried provisioning after erasing your flash using `idf.py erase-flash`  ?

shahpiyushv, 2022-12-19:
@FlynnMa , considering that the process has reached step 4 of provisioning, there should be some prints indicating that some user id and secret key was received from the phone app, followed by SSID and passphrase and then wifi connection success. Thereafter there should be messages indicating claiming/mqtt error or some crash, because of which the step 4 failed. The full logs can help understand that, but we do not see any of this in the logs you shared.

FlynnMa, 2022-12-20:
@shahpiyushv ,

I don't see that kind of error, but do you think bellow warnning might related?

```
W (2679) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
```

shahpiyushv, 2022-12-20:
@FlynnMa , these logs surely are incomplete. Even the timestamps are till 2723, which means 2.7 seconds since bootup. There must be some QR code you scanned, right? I do not see that in the logs here. The phy_init warning you mentioned may result in failure of Wi-Fi connection itself, but we see that your node has gone beyond the "Confirming Wi-Fi Connection" step, which means that the connection itself went through fine.

FlynnMa, 2022-12-20:
Please check out this,
[image]

As you mentioned QR code, I checked my QR setting by idf.py menuconfig, it is enabled. Don't know why it is not showing up.

When I was working with old version, it was fine.

shahpiyushv, 2022-12-20:
@FlynnMa , where exactly did you do the change from `#include <include/esp_wifi.h>` to `#include <esp_wifi.h>`?

When you captured the above log, did  you do a manual provisioning (i.e. without QR code) and provided the pop (a0df7e8b) as mentioned in your log? Did it still proceed till step 4 and yet not print any logs after the efuse logs above? 

Using `git status` command, can you ensure that you have no custom changes? Moreover, can you ensure that all the submodules in esp-idf and esp-rainmaker are in sync using `git submodule update --init --recursive `?

FlynnMa, 2022-12-25:
Today I see this error. it is replicable by my side
[image]

shahpiyushv, 2023-02-16:
@FlynnMa , did this eventually work for you? The latest logs point to some hardware issue rather than a software issue since the device itself has disconnected from the serial terminal. Also, we never really got any logs for the original issue which talks about failure at step 4 of provisioning.