espressif/esp-rainmaker #209: how to create my own rainmaker cloud?

thuanpv1, 2022-12-21:
Hello masters, 

I just want to ask if there is a way to clone ESP rainmaker cloud to my machine and run it localy? I read somewhere on internet that the ESP rainmaker cloud is open source, but I cannot found its source code. Please help.
(this is just a question not an issue, but not found where to change the issue type, so i just submit as an issue)

Regards,
Thuan

sanketwadekar, 2022-12-21:
Hi,
The Rainmaker Agent is open-source, whereas the Rainmaker Cloud is not. It is available to use for free for educational/hobby purposes. You can get in touch with our sales team at [sales@espressif.com](mailto:sales@espressif.com) for details about private deployments.

Thanks

shahpiyushv, 2023-01-13:
Closing this issue because it has been answered.

closed 2023-01-13 by shahpiyushv