espressif/esp-rainmaker #210: Adding Devices

MadanMeelan, 2022-12-21:
How many devices can we add in Rainmaker app for free, I need 8 devices for complete home, can we add all 8 devices through public platform? Please help me on this...

sanketwadekar, 2022-12-21:
Hello,
Public RainMaker has a default quota limit of 5, which we can increase based on user requirements only for hobby/evaluation purposes.
If you have reached your quota, you can get it increased by sending an email to [esp-rainmaker-admin@espressif.com](mailto:esp-rainmaker-admin@espressif.com)

MadanMeelan, 2022-12-21:
Can we increase it to 8 devices? Is there any time limit for the devices which were added in public Rainmaker app?

MadanMeelan, 2022-12-21:
Hello,

How much time it'll take to extend the device count after sending the
request? One more question, is there any duration limit for the devices
which were added to the public Rainmaker app? Please let me know.

Regards
Madan Mohan Reddy

MadanMeelan, 2022-12-21:
I'm practicing it for my house as a hobby, please let me know

shahpiyushv, 2022-12-21:
@MadanMeelan , increasing the quota doesn't take much time. Just send a request to [esp-rainmaker-admin@espressif.com](mailto:esp-rainmaker-admin@espressif.com) as mentioned [above](https://github.com/espressif/esp-rainmaker/issues/210#issuecomment-1360917854). 

Meanwhile, there is no specific expiry for the devices added to public RainMaker.

JohnMacrae, 2022-12-21:
Hi Piyush - season's greetings :)

At this point I think it might be worth clarifying that the limit is on
physical devices (ESP32) rather than the number of (household) devices
driven from one ESP32?

Kind Regards
John

shahpiyushv, 2022-12-21:
@JohnMacrae , thanks for raising this point :). Indeed, this limit is on the number of nodes (physical ESP32s) and not the number of devices. A single ESP32 [node](https://rainmaker.espressif.com/docs/spec-concepts.html#node) can have [multiple devices](https://rainmaker.espressif.com/docs/spec-concepts.html#devices). Moreover, this is applicable only for host/assisted claiming (which are the only options if you are using ESP32 anyways). For ESP32C3, ESP32S2 and ESP32S3, if you use self claiming, then this quota limit does not apply.

MadanMeelan, 2022-12-21:
@Piyush what does it mean Self claiming? As I'm starting please explain

shahpiyushv, 2022-12-21:
@MadanMeelan , you can read more about various types of claiming [here](https://rainmaker.espressif.com/docs/claiming.html).

shahpiyushv, 2023-01-03:
Closing this issue since it has been answered.

closed 2023-01-03 by shahpiyushv