espressif/esp-rainmaker #212: OTA - 2 variants understanding

jacek12345, 2022-12-23:
Sorry for my question if I completely mistanderstood OTA flow.
I'm trying to choose which variant should be used is developed product.
Analysing, but each variant have something that make it unconvenient or even unusable in end product.
Suppose that i misunderstand somwething, so please correct.

1. OTA Using Parameters
"This is the simplest way of pushing out an OTA FW upgrade by "End Users" and can be triggered from the RainMaker CLI."

Great, End User (primary or secondary) is the user that bought the product and controll it via mobile app. He can update firmware in his product, this is what I need. But wait.... he has to use CLI? Not pusch the button in mobile app? Impossible.

2. OTA Using Topics
"This is a more advanced way of pushing out an OTA FW Upgrade, avaiable to "Admin Users" and can be triggered from the RainMaker Dashboard."

Great, so this is what I need. I (developer) can pusch firmware and even decide updating policy and have very usefull dashboard for manage process. But wait... who is Admin User? 
"Admin User" - A user who owns the MQTT credentials of a given node is the Admin User for that node. So, a user who claims a node either using Host driven claiming or Assisted Claiming becomes an admin user for the given node."
So, developer (me) is not Admin User for product (node) that somebody bought, registered and climed withg his app assistance? End customer have to use ESP Insights Dashoard for frimware updating? Impossible. 

I need some simple method for updating firmware by end user, directly from his mobile app (simply pusch the button) - firmware should be pusched on server by developer (me).
OR
I need method for updating all devices that all people ever bought, claimed and use on the whole world. I should see this all in my ESP Insights Dashboard and should start OTA.

What is the best method for supply end product to end user with firmware upgrade possibility (as simple as pusch button for end customer or ALL_DEVICES_UPGRADING available for developer)?

shahpiyushv, 2022-12-27:
@jacek12345 , claiming is a concept only for the public RainMaker deployment. To build a PoC using the public RainMaker, the best approach is to use [Host Driven Claiming](https://rainmaker.espressif.com/docs/claiming.html#host-driven-claiming) with your account so that you become the admin. Then once the nodes are in field, create an OTA Job from dashboard by setting the "User Approval" flag
[image]

Thereafter, there are some `/user/nodes/ota_update` APIs under [OTA Service](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service), which can be called by end users to see if there is any OTA Job available for their nodes and if there is, just start it. The corresponding support is also being added in the RainMaker phone apps.

In private RainMaker deployments, the process of node registration is different and so, the admin ownership is established during the manufacturing step itself.

There are some other features being added to OTA via dashboard. We will improve the documentation so that the usage is easier to understand.

jacek12345, 2022-12-27:
thank You @shahpiyushv 

> In private RainMaker deployments, the process of node registration is different and so, the admin ownership is established during the manufacturing step itself

Do You have any docs with information how to establish the admin ownership during the manufacturing for private RainMaker deployment?
I think, this is to early for private deployment for us but would know how the process looks.

shahpiyushv, 2022-12-27:
@jacek12345, please check out the [Rainmaker Admin CLI](https://github.com/espressif/esp-rainmaker-admin-cli) . This CLI can be used only with private deployments and can't be tested with the public RainMaker.

jacek12345, 2022-12-30:
> Thereafter, there are some /user/nodes/ota_update APIs under [OTA Service](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service), which can be called by end users to see if there is any OTA [...]

Thank You @shahpiyushv 
I'm trying with swagger UI, but i don't know what is "base64_fwimage" in [otaimage](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/useruploadotaimage) command.

jacek12345, 2023-01-02:
> @jacek12345, please check out the [Rainmaker Admin CLI](https://github.com/espressif/esp-rainmaker-admin-cli) . This CLI can be used only with private deployments and can't be tested with the public RainMaker.

There is: "ESP RainMaker Admin CLI is a tool offered by Espressif Rainmaker for admin users to be able to perform mass manufacturing of nodes of ESP32-S2 and ESP32 based products. This tool will enable you to perform node id generation and certificate registration operations required for the manufacturing process." 
What about ESP32-S3?

shahpiyushv, 2023-01-03:
@jacek12345 

> Can You confirm or correct, that can simply use this set of commands from mobile app for OTA?:
> 
> [GET ota_update](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/getotaupdate)
> [POST ota_update](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/postotaupdate)
> [GET ota_status](https://swaggerapis.rainmaker.espressif.com/#/OTA%20Service/getnodeotastatus)
> Or should use something more?

This is correct. You can upload OTA Image and create the OTA Job from the RainMaker dashboard with "User Approval" enabled and thereafter, the above 3 APIs can be used to query and start the OTA Job from the end user side.

> What about ESP32-S3?

All ESP32 based chips are supported. We will rephrase the line soon.

jacek12345, 2023-01-03:
Thank You @shahpiyushv

closed 2023-01-03 by jacek12345