espressif/esp-rainmaker #213: OTA job failed if NimBle enabled

jacek12345, 2022-12-23:
I use ESP32-C3 and trying OTA update.
OTA job ends with success since there is not NimBle initialisation in working code. When i enable NimBle, OTA job fails. I use NimBle for communicate with BLE sensors.
I enable Nimble:
`ESP_ERROR_CHECK(esp_nimble_hci_and_controller_init());
    nimble_port_init();
    //Initialize the NimBLE host configuration.
    ble_hs_cfg.reset_cb = bleprph_on_reset;
    ble_hs_cfg.sync_cb = bleprph_on_sync;
    nimble_port_freertos_init(bleprph_host_task);`

Than while OTA job i have this:
`I (30) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (30) boot: compile time 07:52:06
I (30) boot: chip revision: 3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (41) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (77) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (85) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (92) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (100) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (107) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (115) boot:  7 devdata          WiFi data        01 02 00346000 00003000
I (122) boot:  8 coredump         Unknown data     01 03 00349000 00010000
I (130) boot: End of partition table
I (134) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=3e8b8h (256184) map
I (183) esp_image: segment 1: paddr=0005e8e0 vaddr=3fc94000 size=01738h (  5944) load
I (184) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=106284h (1073796) map
I (356) esp_image: segment 3: paddr=001662ac vaddr=3fc95738 size=016b0h (  5808) load
I (358) esp_image: segment 4: paddr=00167964 vaddr=40380000 size=13fc0h ( 81856) load
I (377) esp_image: segment 5: paddr=0017b92c vaddr=50000010 size=00010h (    16) load
I (383) boot: Loaded app from partition at offset 0x20000
I (384) boot: Disabling RNG early entropy source...
I (397) cpu_start: Pro cpu up.
I (406) cpu_start: Pro cpu start user code
I (406) cpu_start: cpu freq: 160000000
I (406) cpu_start: Application information:
I (409) cpu_start: Project name:     switch
I (414) cpu_start: App version:      2.7
I (418) cpu_start: Compile time:     Dec 23 2022 15:06:08
I (425) cpu_start: ELF file SHA256:  2d2402b581a85610...
I (430) cpu_start: ESP-IDF:          v4.4.3
I (435) heap_init: Initializing. RAM available for dynamic allocation:
I (443) heap_init: At 3FC9E7F0 len 0003DF20 (247 KiB): DRAM
I (449) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (456) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (462) spi_flash: detected chip: generic
I (467) spi_flash: flash io: dio
I (471) sleep: Configure to isolate all GPIO pins in sleep state
I (477) sleep: Enable automatic switching of GPIO sleep configuration
I (484) cpu_start: Starting scheduler.
I (489) esp_rmaker_console: Initialising UART on port 0
I (489) uart: queue free spaces: 8
I (499) esp_rmaker_commands: Registering command: up-time
I (499) esp_rmaker_commands: Registering command: mem-dump
I (509) esp_rmaker_commands: Registering command: task-dump
I (509) esp_rmaker_commands: Registering command: cpu-dump
I (519) esp_rmaker_commands: Registering command: sock-dump
I (529) esp_rmaker_commands: Registering command: heap-trace
I (529) esp_rmaker_commands: Registering command: reset-to-factory
I (539) esp_rmaker_commands: Registering command: add-user
I (549) esp_rmaker_commands: Registering command: get-node-id
I (549) esp_rmaker_commands: Registering command: wifi-prov
I (559) esp_rmaker_commands: Registering command: local-time
I (559) esp_rmaker_commands: Registering command: tz-set
[... 132 lines cut ...]

I (36139) esp_rmaker_ota: Image bytes read: 255265
I (39219) esp_rmaker_ota: Image bytes read: 306465
E (39389) Dynamic Impl: alloc(16749 bytes) failed
E (39389) esp-tls-mbedtls: read error :-0x7F00:
E (39389) TRANSPORT_BASE: esp_tls_conn_read error, errno=Success
E (39389) esp-tls-mbedtls: read error :-0x7200:
E (39399) TRANSPORT_BASE: esp_tls_conn_read error, errno=Success
E (39409) esp_https_ota: data read -1, errno 0
E (39409) esp_rmaker_ota: ESP_HTTPS_OTA upgrade failed ESP_FAIL
I (39419) esp_rmaker_ota: Reporting failed: OTA failed: Error ESP_FAIL
I (39419) esp_rmaker_ota_using_topics: {"ota_job_id":"6DfAeBS7UKJWD4GXRzKoYC","status":"failed","additional_info":"OTA failed: Error ESP_FAIL"}
E (39439) esp_rmaker_ota: Complete data was not received.
I (39439) esp_image: segment 0: paddr=001b0020 vaddr=3c110020 size=3ed70h (257392) map
I (39489) esp_image: segment 1: paddr=001eed98 vaddr=3fc94200 size=01280h (  4736)
I (39489) esp_image: segment 2: paddr=001f0020 vaddr=42000020 size=107c68h (1080424) map
I (39629) esp_image: segment 3: paddr=002f7c90 vaddr=3fc95480 size=01b70h (  7024)
I (39629) esp_image: segment 4: paddr=002f9808 vaddr=40380000 size=1400ch ( 81932)
I (39649) esp_image: segment 5: paddr=0030d81c vaddr=50000010 size=00010h (    16)
E (39649) esp_image: Checksum failed. Calculated 0x48 read 0xe5
E (39649) esp_rmaker_ota: Image validation failed, image is corrupted
I (39659) esp_rmaker_ota: Reporting failed: Image validation failed
I (39659) esp_rmaker_ota_using_topics: {"ota_job_id":"6DfAeBS7UKJWD4GXRzKoYC","status":"failed","additional_info":"Image validation failed"}
I (39979) app_main: MQTT Published. Msg id: 38621.
I (40129) app_main: MQTT Published. Msg id: 42449.`

sanketwadekar, 2022-12-24:
This error `E (39389) esp-tls-mbedtls: read error :-0x7F00:` means that your device has less memory. Events have been added to OTA which allows you to execute your custom code. You can try turning off BLE on `RMAKER_OTA_EVENT_STARTING` event.

jacek12345, 2022-12-31:
Thank You @sanketwadekar 
Done like You said. Also will change module to bigger, since now have 4MB Flash only. Can You clarify if it is due to small FLASH, ROM or SRAM?

sanketwadekar, 2022-12-31:
The error is caused by low “Heap Memory” at runtime, which means that the RAM is insufficient.

jacek12345, 2022-12-31:
Thank You @sanketwadekar

closed 2022-12-31 by jacek12345