espressif/esp-rainmaker #214: Confirming Node Association Failed

anjalir14, 2022-12-24:
Hello

I'm using an ESP32-S3 device. I was trying to connect the device with rainmaker but after scanning the QR code and connecting with wifi the app is unable to confirm the node association and it's due to the MQTT. 

As you mentioned earlier, add an event handler if it's crashing and rebooting but I'm not having any issues like this.

I'm sharing my logs so please once go through them.

[image]

shahpiyushv, 2022-12-27:
@anjalir14 , can you share the logs preceding this, specifically the ones mentioning some claiming related messages and the node id too?

anjalir14, 2022-12-31:
Now, something new thing is happening . Earlier I was using IDF 5.0 version then I downgrade the version to 4.4.3 now I'm not able to generate QR code and there is no error. 
following I have attached all my log.
[image]
[image]
[image]

anjalir14, 2022-12-31:
After trying on version 5 . This is the logs I'm getting

[image]
[image]

anjalir14, 2023-01-05:
Hello,
It's been a long time that i I din't received any answer from your side. so I request you to pls have a look since I'm trying to solve this issue from last 1 week.

Hope for your reply as soon as possible.
thank you

anjalir14, 2023-01-06:
Hello Sir,

I tried exactly what you said to me but it's still not happening.  I'm getting same error. Firstly the QR code was not generating so from some solution I found that I have to make a little change in main.cpp i.e. (app_wifi_start(MAC)instead of RANDOM we have to change MAC ) after this I was able to create QR code otherwise it was rebooting itself again and again.

Below are my logs after rebooting.
W (93562) wifi:<ba-del>idx
W (93562) wifi:<ba-del>idx
I (93562) wifi:new:<11,0>, old:<11,2>, ap:<255,255>, sta:<11,2>, prof:1
I (93572) app_wifi: Disconnected. Connecting to the AP again...
I (93632) wifi:flush txq
I (93632) wifi:stop sw txq
I (93632) wifi:lmac stop hw txq
I (93712) app_main: Rebooting in 2 seconds.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b45
0x40375b45: esp_restart_noos_dig at /home/dnk067/ESP/esp-idf/components/esp_system/esp_system.c:46 (discriminator 1)

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3810,len:0x167c
load:0x403c9700,len:0xbe8
load:0x403cc700,len:0x2ef4
entry 0x403c9904
I (37) boot: ESP-IDF v5.0 2nd stage bootloader
I (37) boot: compile time 12:35:42
I (37) boot: chip revision: v0.1
I (38) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (46) boot.esp32s3: Boot SPI Speed : 80MHz
I (50) boot.esp32s3: SPI Mode       : SLOW READ
I (56) boot.esp32s3: SPI Flash Size : 4MB
I (60) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (84) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (92) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (99) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (106) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (114) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (122) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (129) boot: End of partition table
I (133) boot_comm: chip revision: 1, min. application chip revision: 0
I (141) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=47580h (292224) map
I (218) esp_image: segment 1: paddr=000675a8 vaddr=3fc9ba00 size=04f40h ( 20288) load
I (223) esp_image: segment 2: paddr=0006c4f0 vaddr=40374000 size=03b28h ( 15144) load
I (228) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=e83a8h (951208) map
I (456) esp_image: segment 4: paddr=001583d0 vaddr=40377b28 size=13e98h ( 81560) load
I (478) esp_image: segment 5: paddr=0016c270 vaddr=50000000 size=00010h (    16) load
I (489) boot: Loaded app from partition at offset 0x20000
I (489) boot: Disabling RNG early entropy source...
I (500) cpu_start: Pro cpu up.
I (500) cpu_start: Starting app cpu, entry point is 0x403756a4
0x403756a4: call_start_cpu1 at /home/dnk067/ESP/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (515) cpu_start: Pro cpu start user code
I (515) cpu_start: cpu freq: 160000000 Hz
I (515) cpu_start: Application information:
I (518) cpu_start: Project name:     switch
I (523) cpu_start: App version:      1.0
I (528) cpu_start: Compile time:     Jan  6 2023 12:35:30
I (534) cpu_start: ELF file SHA256:  14f46be02f08a750...
I (540) cpu_start: ESP-IDF:          v5.0
I (545) heap_init: Initializing. RAM available for dynamic allocation:
I (552) heap_init: At 3FCA72C8 len 00042448 (265 KiB): D/IRAM
I (558) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
[... 158 lines cut ...]
I (82067) wifi_prov_mgr: Provisioning stopped
W (82067) BT_INIT: esp_bt_mem_release not implemented, return OK
I (82067) wifi_prov_scheme_ble: BTDM memory released
I (82067) esp_rmaker_local: Event 6
W (82407) wifi:<ba-add>idx:1 (ifx:0, 90:8d:78:76:27:00), tid:1, ssn:0, winSize:64
I (82407) esp_rmaker_time: SNTP Synchronised.
I (82407) esp_rmaker_time: The current time is: Fri Jan  6 15:10:22 2023 +0800[CST], DST: No.
I (82427) HTTP_CLIENT: Body received in fetch header state, 0x3fcf365d, 129
I (82617) esp-x509-crt-bundle: Certificate validated
I (85087) HTTP_CLIENT: Body received in fetch header state, 0x3fcf3406, 502
I (85097) esp_claim: Self Claiming was successful. Certificate received.
I (85097) app_main: RainMaker Claim Successful.
E (85097) esp_rmaker_core: Failed to initialise MQTT Config after claiming. Aborting
I (85107) esp_rmaker_ota_using_topics: Subscribing to: node/68B6B33CD3D8/otaurl
W (85117) esp_rmaker_mqtt: esp_rmaker_mqtt_unsubscribe not registered
W (85127) esp_rmaker_mqtt: esp_rmaker_mqtt_subscribe not registered
W (85127) esp_rmaker_mqtt_budget: MQTT budgeting not started yet. Allowing publish.
W (85137) esp_rmaker_mqtt: esp_rmaker_mqtt_publish not registered
I (85147) esp_rmaker_user_mapping: MQTT Publish: {"node_id":"68B6B33CD3D8","user_id":"Google_BExsnsibETGiNNa8R5qVbz","secret_k
ey":"b49561f1-1422-483d-8755-469b353e6992","reset":true}
I (90127) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
W (90127) esp_rmaker_mqtt_budget: MQTT budgeting not started yet. Allowing publish.
W (90127) esp_rmaker_mqtt: esp_rmaker_mqtt_publish not registered

Waiting for your reply as soon as possible.

shahpiyushv, 2023-01-06:
@anjalir14, your earlier log had Assisted Claiming and the new one has Self Claiming. So I guess you have changed the config and probably also erase the flash, else the claiming wouldn't have started again. As mentioned earlier, can you just reset wifi and retry? Meanwhile, we will check why the device is getting this error `E (85097) esp_rmaker_core: Failed to initialise MQTT Config after claiming. Aborting`. That is causing the failures you see.

anjalir14, 2023-01-07:
Hello @shahpiyushv 
I tried everything that has been mentioned but still MQTT is not intializating. 
Actually I'm running on my deadline of the project, so I would like you to pls check this . Can you pls let me know when this issue is going to solve as my deadline of project is very close.
Hoping to get your answer as soon as possible

sanketwadekar, 2023-01-07:
Hi,
Can you specify your branch and commit hash of  IDF and Rainmaker repos ?

shahpiyushv, 2023-01-11:
@anjalir14 , it would have really helped if logs for the boot cycle after this (with only wifi credentials reset) were shared. Since both the logs are with device booting up in unclaimed state, it is hard to tell why things may not have worked even during the subsequent boot cycles when the node was already claimed. The issue with claiming indeed looks to be some issue, but atleast pre-claimed nodes should work fine.

Meanwhike, as asked above, please share the branch and commit hash of IDF and Rainmaker repos.

closed 2023-02-16 by shahpiyushv