espressif/esp-rainmaker #215: IDF 4.4 compatibility problem

eos1d3, 2022-12-27:
With latest ESP IDF 4.4,  any IDF commands: `set-target`, `build`, `menuconfig`, will cause error. Examples used are `switch` and `led_light`.

```
4 problems were found in the manifest file /Users/andy/esp/esp-rainmaker/components/esp_rainmaker/idf_component.yml:
Invalid manifest format
Invalid dependency format
Dependency version spec format is invalid
Unknown keys in dependency details: rules
CMake Error at /Users/andy/esp/esp-idf/tools/cmake/build.cmake:452 (message):
Call Stack (most recent call first):
  /Users/andy/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:16 (project)

-- Configuring incomplete, errors occurred!
```

shahpiyushv, 2022-12-27:
@eos1d3 , can you upgrade your idf component manager using `pip install --upgrade idf-component-manager` and retry?

eos1d3, 2022-12-27:
Hi, just tested with the above command, it fixes the problem now. Why is the command needed as other IDF versions do not need it?

shahpiyushv, 2022-12-27:
@eos1d3 ESP IDF versions before v4.4 do not use component manager by default and hence, the manifest file gets ignored. idf v5.0 already has the latest component manager python package and so, that too works fine. v4.4 had some changes that caused these failures and they were eventually fixed after the release, hence the issues and the requirement if manually updating it.

eos1d3, 2022-12-28:
Thanks for your help!

closed 2022-12-28 by eos1d3