espressif/esp-rainmaker #216: IDF 5.0 compatibility problem

eos1d3, 2022-12-27:
ESP IDF 5.0, with Rainmaker example: `led_light`

```
In file included from /Users/user1/esp/esp-idf/components/esp-tls/private_include/esp_tls_mbedtls.h:8,
                 from /Users/user1/esp/esp-idf/components/esp-tls/esp_tls_mbedtls.c:16:
/Users/user1/esp/esp-idf/components/esp-tls/esp_tls.h:210:9: error: unknown type name 'mbedtls_ssl_hs_cb_t'
  210 | typedef mbedtls_ssl_hs_cb_t esp_tls_handshake_callback;
      |         ^~~~~~~~~~~~~~~~~~~
[554/1004] Building C object esp-idf/http_parser/CMakeFiles/__idf_http_parser.dir/http_parser.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/user1/esp/esp-rainmaker/examples/led_light/build/log/idf_py_stderr_output_66186 and /Users/user1/esp/esp-rainmaker/examples/led_light/build/log/idf_py_stdout_output_66186
```

shahpiyushv, 2022-12-27:
@eos1d3 The issues are seen because some incompatible changes happened in esp-rainmaker and esp-idf, especially related to the commit [examples: Disable mbedTLS server config to reduce firmware size
](https://github.com/espressif/esp-rainmaker/commit/7c1e894fe028a1094b99dea098d62764cdbd9e58)

You can either remove `CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y`  from your sdkconfig or revert to older config using:

idf.py menuconfig -> Component config -> mbedTLS -> TLS Protocol Role -> Server & Client

This will be fixed cleanly in esp idf v5.0.1 and a future commit in esp-rainmaker.

MrRedKite, 2023-08-15:
@mik13ST could you solve the problem?
Did it the same way you did, but removed `CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y` and used `CONFIG_MBEDTLS_TLS_SERVER_AND_CLIENT=y` instead, like @shahpiyushv wrote.
But i still get the same error -.-

mik13ST, 2023-08-15:
@MrRedKite No, I didn't. In my case migrating to wolfSSL was just an experiment so I didn't have the motivation to get it working.

shahpiyushv, 2023-08-18:
@MrRedKite , can you file a new issue for your observations? Closing this issue for the time being since esp-rainmaker now works well with release/v5.0 and release/v5.1 branches too.

closed 2023-08-18 by shahpiyushv

MrRedKite, 2023-08-18:
@shahpiyushv, I switched back to mbedtls, so I dont need a new issue.