espressif/esp-rainmaker #217: ESP Insights: "heap_metrics: Failed to add heap metric key:free"

jacek12345, 2022-12-28:
I turned on ESP Insights with heap and wifi metrics and after some time it gets warnings.
It was one time issue, after restart problem not occured yet. Do You know what it can be related?
I enabled also Component config->App Insights ->'Enable all diagnostics log type'  for WARNING presentation.
Using ESP32-C3-DevKitM-1, IDF v.4.4.3

`I (30878) heap_metrics: free:0xccfc lfb:0x9400 min_free_ever:0xbb1c
I (30878) wifi_metrics: Wi-Fi RSSI crossed threshold -54
I (30878) wifi_metrics: rssi:-54 min_rssi_ever:-54
I (60878) heap_metrics: free:0xcc14 lfb:0x9400 min_free_ever:0xbb1c
I (60878) wifi_metrics: rssi:-53 min_rssi_ever:-54
I (90878) heap_metrics: free:0xcc94 lfb:0x9400 min_free_ever:0xb45c
I (90878) wifi_metrics: rssi:-54 min_rssi_ever:-54
I (120878) heap_metrics: free:0xcb5c lfb:0x9400 min_free_ever:0xb45c
I (120878) wifi_metrics: rssi:-51 min_rssi_ever:-54
I (150878) heap_metrics: free:0xcc7c lfb:0x9400 min_free_ever:0xb45c
I (150878) wifi_metrics: rssi:-53 min_rssi_ever:-54
I (180878) heap_metrics: free:0xcc94 lfb:0x9400 min_free_ever:0xb45c
I (180878) wifi_metrics: rssi:-53 min_rssi_ever:-54
I (210878) heap_metrics: free:0xcc94 lfb:0x9400 min_free_ever:0xb45c
I (210878) wifi_metrics: rssi:-51 min_rssi_ever:-54
I (240878) heap_metrics: free:0xcc94 lfb:0x9400 min_free_ever:0xb45c
I (240878) wifi_metrics: rssi:-52 min_rssi_ever:-54
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (270878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (270878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (300878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (300878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (330878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
I (330878) wifi_metrics: Wi-Fi RSSI crossed threshold -56
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (330898) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (360878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (360878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (390878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (390878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (420878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (420878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (450878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (450878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (480878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (480878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (510878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (510878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (540878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (540878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
I (550078) wifi_metrics: Wi-Fi RSSI crossed threshold -57
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
I (552428) wifi_metrics: Wi-Fi RSSI crossed threshold -58
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (570878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (570878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (600878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (600878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (630878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (630878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (660878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
W (660878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
W (690878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
rtc_store_critical_data_write failed len 120, err 0x0101
W (690878) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
rtc_store_critical_data_write failed len 120, err 0x0101
W (720878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
rtc_store_critical_data_write failed len 120, err 0x0101
W (720888) wifi_metrics: Failed to add Wi-Fi metrics key:rssi
rtc_store_non_critical_data_write failed group heap, len 40, err 0x0101
rtc_store_critical_data_write failed len 120, err 0x0101
W (750878) heap_metrics: Failed to add heap metric key:free
rtc_store_non_critical_data_write failed group wifi, len 40, err 0x0101
rtc_store_critical_data_write failed len 120, err 0x0101
W (750888) wifi_metrics: Failed to add Wi-Fi metrics key:rssi`

vikramdattu, 2023-01-02:
Hello @jacek12345 above errors appear when the RTC store (we record all the insights messages here till they are sent to the cloud) is full.

Here are the few steps to circumvent the issue:
1. Disable system metrics which are not useful to you
2. Do concise insights logging if you are doing it

You are seeing the device gets away with above logs after hard reset, because hard reset causes RTC data invalid and empties the store. Hope this helps.

jacek12345, 2023-01-02:
Thank You @vikramdattu

closed 2023-01-02 by jacek12345