espressif/esp-rainmaker #218: button_handle_t takes only Active_Low Buttons

exb53, 2022-12-31:
I am not very experienced in programming the ESP32 with Rainmaker. 
In the gpio_button component seems to be a bug. It works only with active low buttons. So I did a little change in button.c. 
I replaced in lines 248/249

    gpio_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
    gpio_conf.pull_up_en = GPIO_PULLUP_ENABLE;

with:
    if (active_level == BUTTON_ACTIVE_LOW)
    {
        gpio_conf.pull_down_en = GPIO_PULLDOWN_DISABLE;
        gpio_conf.pull_up_en = GPIO_PULLUP_ENABLE;
    }
    else
    {
        gpio_conf.pull_down_en = GPIO_PULLDOWN_ENABLE;
        gpio_conf.pull_up_en = GPIO_PULLUP_DISABLE;
    }
That works on both active low and active high button designs for me.

Please let me know, if there is any issue with that.

Thank you

shahpiyushv, 2023-01-03:
@exb53 , This indeed seems to be a bug. However, this `gpio_button  component will soon be deprecated as the same functionality is available in the updated [button component](https://github.com/espressif/esp-iot-solution/tree/master/components/button). I see that it already has that fix [here](https://github.com/espressif/esp-iot-solution/blob/master/components/button/button_gpio.c#L37).

exb53, 2023-01-03:
Thank you, as I said I am not too experienced in programming . So should I delete the buttom‘s rainmaker include and use the iot‘s instead until a fixed rainmaker lib is available? Or should I stick with my fix?