espressif/esp-rainmaker #219: Problem with assisted claiming

jacek12345, 2023-01-02:
Is it normal that each started claiming changes NodeID?
[image]

Is this NodeID changing the reason of Claiming abort "User does not have available quota"?

The problem started today after added to my code:
- erasing of user data nvs partition and factory reset:
err = nvs_flash_erase_partition_ptr("devdata");
factory_reset_trigger(NULL);

- added some data to this partition
- made "erase-flash"
After that, my iOS app shows something with BLE and suggest to restart device
After that, there is  "User does not have available quota" in app
Also see that each try changes NodeID.

shahpiyushv, 2023-01-02:
@jacek12345 , with assisted claiming, for the same mac address + user id combination, the same node id gets assigned. However, the code flow is such that it first reads the node id (which is equal to mac address if a valid one is not found in nvs) and then that is used for the claiming. In your case, some node id is found in nvs and so that is getting used for claiming instead of the mac address, resulting into a new node id getting assigned. Erasing the fctry partition should resolve this issue.

Meanwhile, recently we made a change such that even for self claiming, the primary end user is made the admin. So, you can also use self claiming.

Your understanding about the private deployment is correct. There is no claiming, but the node registration process ensures that you have admin access to the node.

jacek12345, 2023-01-03:
> Meanwhile, recently we made a change such that even for self claiming, the primary end user is made the admin. So, you can also use self claiming.

Do You mean that it is possible now to use Self Claiming and see node in dashboards?

shahpiyushv, 2023-01-03:
> Do You mean that it is possible now to use Self Claiming and see node in dashboards?

Yes

jacek12345, 2023-01-03:
1. Indeed, I see data coming from node in ESP Insights Dashboard but node status on the node list is "unknown".
2. I can't see node in ESP RainMaker Dashboard

shahpiyushv, 2023-01-13:
@jacek12345 , can you share the node id for this latest issue you mentioned about node visible in Insights, but not on RainMaker dashboard?

jacek12345, 2023-01-13:
it is node: A0764E5AE100. It is ESP32-C3. I see this with unknown state but data comes:
[image]
[image]

Can't see in RainMaker dashboard:
[image]

Now I use ESP32-S3 module and can't see node in Insights nor RainMaker dashboard. Is something more to do with S3 module?

jacek12345, 2023-01-16:
Still can't see my S3 device in RainMaker dashboard nor ESP Insights dashboard.

```
I (24) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (25) boot: compile time 08:54:37
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (79) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 003c0000
I (94) boot:  5 ota_1            OTA app          00 11 003e0000 003c0000
I (102) boot:  6 fctry            WiFi data        01 02 007a0000 00006000
I (109) boot:  7 devdata          WiFi data        01 02 007a6000 00003000
I (117) boot:  8 coredump         Unknown data     01 03 007a9000 00010000
I (124) boot: End of partition table
I (129) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=3ed3ch (257340) map
I (183) esp_image: segment 1: paddr=0005ed64 vaddr=3fc9a180 size=012b4h (  4788) load
I (185) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=eae84h (962180) map
I (361) esp_image: segment 3: paddr=0014aeac vaddr=3fc9b434 size=03818h ( 14360) load
I (365) esp_image: segment 4: paddr=0014e6cc vaddr=40374000 size=1617ch ( 90492) load
I (387) esp_image: segment 5: paddr=00164850 vaddr=50000000 size=00010h (    16) load
I (396) boot: Loaded app from partition at offset 0x20000
I (396) boot: Disabling RNG early entropy source...
I (408) cpu_start: Pro cpu up.
I (408) cpu_start: Starting app cpu, entry point is 0x40375444
0x40375444: call_start_cpu1 at F:/GitHub/esp-idf-v4.4.3/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (422) cpu_start: Pro cpu start user code
I (422) cpu_start: cpu freq: 160000000
I (422) cpu_start: Application information:
I (425) cpu_start: Project name:     proiot1
I (430) cpu_start: App version:      1.1
I (434) cpu_start: Compile time:     Jan 13 2023 10:36:58
I (440) cpu_start: ELF file SHA256:  d19723f5b3afc189...
I (446) cpu_start: ESP-IDF:          v4.4.3
I (451) heap_init: Initializing. RAM available for dynamic allocation:
I (458) heap_init: At 3FCA63F8 len 00043318 (268 KiB): D/IRAM
I (465) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (472) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (478) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (485) spi_flash: detected chip: generic
I (489) spi_flash: flash io: dio
I (493) sleep: Configure to isolate all GPIO pins in sleep state
I (500) sleep: Enable automatic switching of GPIO sleep configuration
I (507) coexist: coexist rom version e7ae62f
I (512) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) esp_rmaker_console: Initialising UART on port 0
I (532) esp_rmaker_commands: Registering command: up-time
I (532) uart: queue free spaces: 8
I (542) esp_rmaker_commands: Registering command: mem-dump
I (552) esp_rmaker_commands: Registering command: task-dump
I (552) esp_rmaker_commands: Registering command: cpu-dump
I (562) esp_rmaker_commands: Registering command: sock-dump
I (572) esp_rmaker_commands: Registering command: heap-trace
I (572) esp_rmaker_commands: Registering command: reset-to-factory
I (582) esp_rmaker_commands: Registering command: add-user
I (592) esp_rmaker_commands: Registering command: get-node-id
I (592) esp_rmaker_commands: Registering command: wifi-prov
I (602) esp_rmaker_commands: Registering command: local-time
I (602) esp_rmaker_commands: Registering command: tz-set
I (612) esp_rmaker_commands: Registering command: cmd
[... 82 lines cut ...]
I (2292) esp_netif_handlers: sta ip: 10.10.10.134, mask: 255.255.255.0, gw: 10.10.10.1
I (2292) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 1
I (2322) esp_https_server: Starting server
I (2322) esp_https_server: Server listening on port 8080
I (2322) esp_rmaker_local: esp_local_ctrl service started with name : 7CDFA1E25EA0
W (2332) app_main: Unhandled RainMaker Event: 6
I (2342) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (2342) esp_rmaker_core: Waiting for MQTT connection
I (2822) esp-x509-crt-bundle: Certificate validated
I (3232) esp_rmaker_time: SNTP Synchronised.
I (3232) esp_rmaker_time: The current time is: Mon Jan 16 08:26:18 2023 +0100[CET], DST: No.
I (4472) esp_mqtt_glue: MQTT Connected
I (4472) app_main: MQTT Connected.
I (4472) esp_rmaker_node_config: Reporting Node Configuration of length 5308 bytes.
I (4792) esp_rmaker_param: Params MQTT Init done.
I (4792) esp_rmaker_param: Reporting params (init): {"Brama":{"Name":"OUT1","Stan":0},"Furtka":{"Name":"OUT2","Stan":0,"Czas [s]":3},"Przycisk":{"Name":"INPUT","Stan":"ON","Min. czas pobudzenia [s]":0,"Powiadom gdy aktywne":false,"Powiadom gdy nieaktywne":false,"Odwroc logike":false},"MENU_SCAN":{"Wykryj nowe czujniki":false,"Usun wszystkie czujniki":false},"MENU_NODE":{"Nazwa urzadzenia":"Receiver","Ukryj komponent OUT 1":false,"Ukryj komponent OUT 2":false,"Ukryj komponent IN 1":false,"Zezw. użytk. na edycję":false},"SSDB6B0C232F46":{"Name":"Sensor(2F46)","Stan":"--","Wersja  [... cut ...] b167c722","Type":1}}
I (4882) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (4882) esp_rmaker_ota_using_topics: Subscribing to: node/7CDFA1E25EA0/otaurl
I (4892) esp_insights: Scheduling Insights timer for 60 seconds.
I (4992) app_main: MQTT Published. Msg id: 2812.
I (5392) app_main: MQTT Published. Msg id: 23420.
I (5392) app_main: MQTT Published. Msg id: 18803.
I (9892) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (10182) app_main: MQTT Published. Msg id: 62926.
```

sanketwadekar, 2023-01-16:
> Still can't see my S3 device in RainMaker dashboard nor ESP Insights dashboard. Using SelfClaiming.
> 
> ```
> I (24) boot: ESP-IDF v4.4.3 2nd stage bootloader
> I (25) boot: compile time 08:54:37
> I (25) boot: chip revision: 0
> I (26) boot.esp32s3: Boot SPI Speed : 80MHz
> I (31) boot.esp32s3: SPI Mode       : DIO
> I (36) boot.esp32s3: SPI Flash Size : 8MB
> I (40) boot: Enabling RNG early entropy source...
> I (46) boot: Partition Table:
> I (49) boot: ## Label            Usage          Type ST Offset   Length
> I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
> I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
> I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
> I (79) boot:  3 phy_init         RF data          01 01 00018000 00001000
> I (87) boot:  4 ota_0            OTA app          00 10 00020000 003c0000
> I (94) boot:  5 ota_1            OTA app          00 11 003e0000 003c0000
> I (102) boot:  6 fctry            WiFi data        01 02 007a0000 00006000
> I (109) boot:  7 devdata          WiFi data        01 02 007a6000 00003000
> I (117) boot:  8 coredump         Unknown data     01 03 007a9000 00010000
> I (124) boot: End of partition table
> I (129) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=3ed3ch (257340) map
> I (183) esp_image: segment 1: paddr=0005ed64 vaddr=3fc9a180 size=012b4h (  4788) load
> I (185) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=eae84h (962180) map
> I (361) esp_image: segment 3: paddr=0014aeac vaddr=3fc9b434 size=03818h ( 14360) load
> I (365) esp_image: segment 4: paddr=0014e6cc vaddr=40374000 size=1617ch ( 90492) load
> I (387) esp_image: segment 5: paddr=00164850 vaddr=50000000 size=00010h (    16) load
> I (396) boot: Loaded app from partition at offset 0x20000
> I (396) boot: Disabling RNG early entropy source...
> I (408) cpu_start: Pro cpu up.
> I (408) cpu_start: Starting app cpu, entry point is 0x40375444
> 0x40375444: call_start_cpu1 at F:/GitHub/esp-idf-v4.4.3/components/esp_system/port/cpu_start.c:148
> 
> I (0) cpu_start: App cpu up.
> I (422) cpu_start: Pro cpu start user code
> I (422) cpu_start: cpu freq: 160000000
> I (422) cpu_start: Application information:
> I (425) cpu_start: Project name:     proiot1
> I (430) cpu_start: App version:      1.1
> I (434) cpu_start: Compile time:     Jan 13 2023 10:36:58
> I (440) cpu_start: ELF file SHA256:  d19723f5b3afc189...
> I (446) cpu_start: ESP-IDF:          v4.4.3
> I (451) heap_init: Initializing. RAM available for dynamic allocation:
> I (458) heap_init: At 3FCA63F8 len 00043318 (268 KiB): D/IRAM
> I (465) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
> I (472) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
> I (478) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
> I (485) spi_flash: detected chip: generic
> I (489) spi_flash: flash io: dio
> I (493) sleep: Configure to isolate all GPIO pins in sleep state
> I (500) sleep: Enable automatic switching of GPIO sleep configuration
> I (507) coexist: coexist rom version e7ae62f
> I (512) cpu_start: Starting scheduler on PRO CPU.
> I (0) cpu_start: Starting scheduler on APP CPU.
> I (10) esp_rmaker_console: Initialising UART on port 0
> I (532) esp_rmaker_commands: Registering command: up-time
> I (532) uart: queue free spaces: 8
> I (542) esp_rmaker_commands: Registering command: mem-dump
> I (552) esp_rmaker_commands: Registering command: task-dump
> I (552) esp_rmaker_commands: Registering command: cpu-dump
> I (562) esp_rmaker_commands: Registering command: sock-dump
> I (572) esp_rmaker_commands: Registering command: heap-trace
> I (572) esp_rmaker_commands: Registering command: reset-to-factory
> I (582) esp_rmaker_commands: Registering command: add-user
> I (592) esp_rmaker_commands: Registering command: get-node-id
> I (592) esp_rmaker_commands: Registering command: wifi-prov
> I (602) esp_rmaker_commands: Registering command: local-time
> I (602) esp_rmaker_commands: Registering command: tz-set
> I (612) esp_rmaker_commands: Registering command: cmd
[... 84 lines cut ...]
> I (2322) esp_https_server: Starting server
> I (2322) esp_https_server: Server listening on port 8080
> I (2322) esp_rmaker_local: esp_local_ctrl service started with name : 7CDFA1E25EA0
> W (2332) app_main: Unhandled RainMaker Event: 6
> I (2342) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
> I (2342) esp_rmaker_core: Waiting for MQTT connection
> I (2822) esp-x509-crt-bundle: Certificate validated
> I (3232) esp_rmaker_time: SNTP Synchronised.
> I (3232) esp_rmaker_time: The current time is: Mon Jan 16 08:26:18 2023 +0100[CET], DST: No.
> I (4472) esp_mqtt_glue: MQTT Connected
> I (4472) app_main: MQTT Connected.
> I (4472) esp_rmaker_node_config: Reporting Node Configuration of length 5308 bytes.
> I (4792) esp_rmaker_param: Params MQTT Init done.
> I (4792) esp_rmaker_param: Reporting params (init): {"Brama":{"Name":"OUT1","Stan":0},"Furtka":{"Name":"OUT2","Stan":0,"Czas [s]":3},"Przycisk":{"Name":"INPUT","Stan":"ON","Min. czas pobudzenia [s]":0,"Powiadom gdy aktywne":false,"Powiadom gdy nieaktywne":false,"Odwroc logike":false},"MENU_SCAN":{"Wykryj nowe czujniki":false,"Usun wszystkie czujniki":false},"MENU_NODE":{"Nazwa urzadzenia":"Receiver","Ukryj komponent OUT 1":false,"Ukryj komponent OUT 2":false,"Ukryj komponent IN 1":false,"Zezw. użytk. na edycję":false},"SSDB6B0C232F46":{"Name":"Sensor(2F46)","Stan":"--","Wers [... cut ...]
> I (4882) esp_rmaker_cmd_resp: Enabling Command-Response Module.
> I (4882) esp_rmaker_ota_using_topics: Subscribing to: node/7CDFA1E25EA0/otaurl
> I (4892) esp_insights: Scheduling Insights timer for 60 seconds.
> I (4992) app_main: MQTT Published. Msg id: 2812.
> I (5392) app_main: MQTT Published. Msg id: 23420.
> I (5392) app_main: MQTT Published. Msg id: 18803.
> I (9892) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
> I (10182) app_main: MQTT Published. Msg id: 62926.
> ```

Do you see the device in the iOS / Android Rainmaker app?

jacek12345, 2023-01-16:
Yes, I see.

jacek12345, 2023-01-16:
This is some problem with my account i think.
I have two accounts for testing owner and user role app on two phones.
When I registered node (owned) on second account and logged in to Rainmaker dashboard and Insights dashboard, i see node (even in SELF CLAIMING). When back to first account and own node and logged in to dashboards, i can't see node.
More, this node is still visible on second account, where it was owned earlier. It is possible that i owned it on second account some time ago and that is why didn't see it on main account for some time. 
Shouldn't be node connected to ACTUAL owner account dashboards?

shahpiyushv, 2023-01-16:
@jacek12345 , from your description, it seems there could indeed be an issue with the logic of primary user being assigned as admin for self claimed nodes. We will check this internally. We have also increased your quota as per the email. Sorry for the delay.

closed 2023-02-18 by jacek12345