espressif/esp-rainmaker #220: Rainmaker SDK building error!

MahendraDnkTech, 2023-01-04:
Hello Geeks 
I hope all are doing well...
I have just configured the ESP32-IDF & rainmaker 
ESP32 IDF is successfully built while the rainmaker is causing the issue during the building!

I have followed the steps mentioned into the rainmaker getting started guide 

The error logs are attached here 
The error comes from the "esp-idf/esp_https_server" directory 

Pl. take a look in to it 

Thanks 

[err_logs.txt](https://github.com/espressif/esp-rainmaker/files/10341919/err_logs.txt)

sanketwadekar, 2023-01-04:
Hi,
It seems like you are using IDF v5. Some changes have been made in rainmaker and IDF which is causing this error (this will be fixed soon). You can compile by using "release/v4.4" branch of IDF.

MahendraDnkTech, 2023-01-04:
Oh! got it @sanketwadekar 
Many thanks 

Let me downgrade the version

Regards
MS

MahendraDnkTech, 2023-01-04:
it's resolved by donwgrading the IDF version to v4.4

Thanks

closed 2023-01-04 by MahendraDnkTech