espressif/esp-rainmaker #221: Assisted claiming issue & Failed to initialise MQTT Config

MahendraDnkTech, 2023-01-04:
Hello Geeks...
Hope all are doing well 

I'm playing an amount ESP32S3 chip & rainmaker SDK 
I have successfully configured & install ESP32-IDF (v4.4) & Rainmaker SDK 
The build also seems good for both SDK 

I have flashed switch example on s3 chip & able to send the WiFi hotspot SSID & password via the rainmaker Android app 
The S3 is also connecting with the WiFi router & got the IP 

After some time during the claiming it's generating the error & claiming got failed 
here are the Error logs

**E (95615) esp_rmaker_core: Failed to initialise MQTT Config after claiming. Aborting**

**E (98525) protocomm_nimble: Error setting advertisement data; rc = 30**

I have attached the full logs here in the text file 

For further debugging, I have logged in with the ESP-Insights dashboard with the same credentials which I was using with the app 
And, I got this message on the dashboard 

All comments & Suggestions Welcome :)

Thanks 
MS

[image]

[esp_rainmaker_err.txt](https://github.com/espressif/esp-rainmaker/files/10344240/esp_rainmaker_err.txt)

MahendraDnkTech, 2023-01-04:
FYI: I have made the 1 change in main_app.c as the code was not generating the QR code before 
here are the changes 

    /* Start the Wi-Fi.
     * If the node is provisioned, it will start connection attempts,
     * else, it will start Wi-Fi provisioning. The function will return
     * after a connection has been successfully established
     */
    err = app_wifi_start(POP_TYPE_MAC);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Could not start Wifi. Aborting!!!");
        vTaskDelay(5000/portTICK_PERIOD_MS);
        abort();
    }

MahendraDnkTech, 2023-01-05:
Hi.. Guys 
Here are the furthermore logs by enabling the debug 
[esp32_rainamekr_err_logs.txt](https://github.com/espressif/esp-rainmaker/files/10350119/esp32_rainamekr_err_logs.txt)

Thanks

sanketwadekar, 2023-01-05:
This seems similar to #214

sanketwadekar, 2023-01-08:
Hi,
Can you revert the code changes you have done and post the log file with the default log level set to "Debug" (using `idf.py menuconfig`) for investigating why the QR Code was not printing?

MahendraDnkTech, 2023-01-09:
Yes, Will do that

Thanks
Mahendra

sanketwadekar, 2023-01-10:
From the logs, it looks like the issue is coming from nvs read operations.
Can you try the nvs_rw_value and nvs_rw_blob examples from IDF to confirm the source of this issue?
Also, try with IDF `release/v5.0` branch as well.

MahendraDnkTech, 2023-01-10:
Yes, that's right 
The issue is coming from the NVS read operation, however, from the logs, you can check it has been detected!
Make sure that my board version is: ESP32-S3-DevKitC-1-N32R8V 

Yes, will perform the test and update the results over here 

Thanks 
MS

MahendraDnkTech, 2023-01-11:
Hi.. @sanketwadekar 

Pl. find the test logs here 
I have  used v4.4 IDF version 

[nvs_rw_value_logs.txt]
[nvs_rw_blob_logs.txt](https://github.com/espressif/esp-rainmaker/files/10392189/nvs_rw_blob_logs.txt)
(https://github.com/espressif/esp-rainmaker/files/10392184/nvs_rw_value_logs.txt)

Thanks 
MS

shahpiyushv, 2023-01-13:
@MahendraDnkTech Can you enable Octal Flash using idf.py menuconfig -> Serial flasher config ->  Enable Octal Flash and retry?

MahendraDnkTech, 2023-01-13:
Hi.. @shahpiyushv 

here are the logs with enabled octa SPI Flash 

[nvs_rw_blob_logs.txt](https://github.com/espressif/esp-rainmaker/files/10410297/nvs_rw_blob_logs.txt)
[nvs_rw_value_logs.txt](https://github.com/espressif/esp-rainmaker/files/10410298/nvs_rw_value_logs.txt)

Thanks

MahendraDnkTech, 2023-01-13:
Hi.. @shahpiyushv 

The switch example of the rainmaker is now working after enabling the Octa spi flash!

Many Thanks