espressif/esp-rainmaker #223: Writing to esp rain maker got error in serial monitor, need a suggestion to solve this issue

Vignesh1006, 2023-01-11:
11:15:27.025 -> E esp_rmaker_param: Current time not yet available. Cannot report time series data.
11:15:36.992 -> E (44667) MQTT_CLIENT: Writing didn't complete in specified timeout: errno=0
11:15:37.024 -> E (44667) esp_mqtt_glue: MQTT_EVENT_ERROR
11:15:37.024 -> E (44668) esp_mqtt_glue: MQTT Publish failed

sanketwadekar, 2023-01-11:
@Vignesh1006  Please mention a detail description of what you are trying to build (which example you have flashed ), the device you are using, and the **complete** debug logs.

Vignesh1006, 2023-01-11:
i've trying to monitor gas leakage using MQ2 with realtime reading on esp rainmaker app, with esp32

using below example

#include <WiFi.h>
#include <RMaker.h>
#include <RMakerDevice.h>
#include <RMakerNode.h>
#include <RMakerParam.h>
#include <RMakerQR.h>
#include <RMakerType.h>
#include "WiFiProv.h"
#include <SimpleTimer.h>
#include <MQ2.h>

const char *service_name = "Vignesh";
const char *pop = "12345678";

// define the Chip Id
uint32_t espChipId = 5;

// define the Node Name
char nodeName[] = "Vignesh";

// define the Device Names

char deviceName_1[] = "LPG";

static uint8_t wifiLed      = 2;   //D2
static uint8_t gpio_reset   = 0;
int MQ2_PIN   = 34;  //RX2
int GREEN_LED   =  14;
int RED_LED      = 25;
int sensorThres = 400;

#define DEFAULT_LPG 0

float lpg;
MQ2 mq2(MQ2_PIN);
SimpleTimer Timer;

static TemperatureSensor LPG("LPG");

void initWiFi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(service_name, pop);
  Serial.print("Connecting to WiFi ..");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println(WiFi.localIP());
}

bool GAS_STATE = false;
uint32_t chipId = 0;

void sysProvEvent(arduino_event_t *sys_event)
{
    switch (sys_event->event_id) {      
        case ARDUINO_EVENT_PROV_START:
#if CONFIG_IDF_TARGET_ESP32
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on BLE\n", service_name, pop);
        printQR(service_name, pop, "ble");
#else
        Serial.printf("\nProvisioning Started with name \"%s\" and PoP \"%s\" on SoftAP\n", service_name, pop);
        printQR(service_name, pop, "softap");
#endif        
        break;
        case ARDUINO_EVENT_WIFI_STA_CONNECTED:
        Serial.printf("\nConnected to Wi-Fi!\n");
        digitalWrite(wifiLed, true);
        break;
    }
}
void write_callback(Device *device, Param *param, const param_val_t val, void *priv_data, write_ctx_t *ctx)
{
    const char *device_name = device->getDeviceName();
    const char *param_name = param->getParamName();
if (strcmp(param_name, "LPG L") == 0) {
Serial.printf("Received value = %2.2f for %s - %s\n", val.val.f, device_name, param_name);
GAS_STATE = val.val.f;
param->updateAndReport(val);
}
  
}
    

void setup()
{
    
    Serial.begin(115200);
    mq2.begin();
     
    pinMode(wifiLed, OUTPUT);
    pinMode(gpio_reset, INPUT);
    pinMode(MQ2_PIN, INPUT);
    pinMode(GREEN_LED, OUTPUT);
  pinMode(RED_LED, OUTPUT);
    digitalWrite(wifiLed, LOW);

    Node my_node;    
    my_node = RMaker.initNode("WhiteFlame");

    LPG.addCb(write_callback);    
my_node.addDevice(LPG);
    

    Timer.setInterval(2000); 

    //This is optional 
    RMaker.enableOTA(OTA_USING_PARAMS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone(). 
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();
    RMaker.enableSchedule();

    //Service Name
    for(int i=0; i<17; i=i+8) {
      espChipId |= ((ESP.getEfuseMac() >> (40 - i)) & 0xff) << i;
    }

    Serial.printf("\nChip ID:  %d Service Name: %s\n", espChipId, service_name);

    Serial.printf("\nStarting ESP-RainMaker\n");
    RMaker.start();

    WiFi.onEvent(sysProvEvent);
#if CONFIG_IDF_TARGET_ESP32
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
#else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
#endif

    LPG.updateAndReportParam(ESP_RMAKER_DEF_POWER_NAME, false);
}

void loop()
{
    // Read GPIO0 (external button to reset device
    if(digitalRead(gpio_reset) == LOW) { //Push button pressed
        Serial.printf("Reset Button Pressed!\n");
        // Key debounce handling
        delay(100);
        int startTime = millis();
        while(digitalRead(gpio_reset) == LOW) delay(50);
        int endTime = millis();

        if ((endTime - startTime) > 10000) {
          // If key pressed for more than 10secs, reset all
          Serial.printf("Reset to factory.\n");
          RMakerFactoryReset(2);
        } else if ((endTime - startTime) > 3000) {
          Serial.printf("Reset Wi-Fi.\n");
          // If key pressed for more than 3secs, but less than 10, reset Wi-Fi
          RMakerWiFiReset(2);
        }
          delay(100);
    }
  

    if (WiFi.status() != WL_CONNECTED)
    {
      //Serial.println("WiFi Not Connected");
      digitalWrite(wifiLed, false);
    }
    else
    {
      //Serial.println("WiFi Connected");
      digitalWrite(wifiLed, true);
      if (Timer.isReady()) { 
        //Serial.println("Sending Sensor Data");
        
        Timer.reset();      // Reset a second timer
      }
    }
 ReadMQ2Sensor();
   SendMQ2Sensor();   

}
void ReadMQ2Sensor()
     {
       int analogSensor = analogRead(MQ2_PIN);
  
  if (analogSensor > sensorThres)
  {
   Serial.print("Pin A0: ");
    Serial.println("Gas Leakage is detected!");
  
    digitalWrite(RED_LED, HIGH);
    digitalWrite(GREEN_LED, LOW);  
   }
  else
  {
    digitalWrite(RED_LED, LOW);
    digitalWrite(GREEN_LED, HIGH);
    
   }
  delay(100);
}
void SendMQ2Sensor()
{
   float* values= mq2.read(true); //set it false if you don't want to print the values to the Serial
float t = mq2.readLPG();
  Serial.print("LPG - "); Serial.println(t);
  LPG.updateAndReportParam("Temperature",t);
    if(t>250) {
      Serial.println("LPG above 250!");
     esp_rmaker_raise_alert("Security Alert!\nGas Leakage is detected."); 
      //digitalWrite(BUZZER_PIN, HIGH);
      //BUZZER_STATE = true;
      //buzzer_timer = millis();
      delay(100);
    }
  
}

sanketwadekar, 2023-01-11:
@Vignesh1006  Increase the timer interval to 5 seconds at least and send sensor readings only if WiFi is connected and the sensor reading has changed from its previous value.

Vignesh1006, 2023-01-11:
ok, will modify it
thanks for your suggestion.

Vignesh1006, 2023-01-12:
its also failed, actually it works fine only for 2 to 3 cycles, then the data not received to esp rainmaker, values are got stuck in ui and alert also not raised in notification.

Vignesh1006, 2023-01-27:
Thanks dude

Vignesh1006, 2023-01-28:
How to read ultrasonic sensor data on esp rainmaker using esp32 or else how to send serial monitor values to esprainmaker

Vignesh1006, 2023-02-04:
Actually, it was works fine after set time interval to 6 sec

closed 2023-02-05 by Vignesh1006