espressif/esp-rainmaker #226: esp_rmaker_param_add_valid_str_list() - possible list update?

jacek12345, 2023-01-17:
Is it possible to update valid string list after it was added?
When is the list sended to server? I cant't see it in monitor.

sanketwadekar, 2023-01-17:
Yes, it is possible. If you want to add/remove any devices and want the changes to get reflected in the phone app, then you need to call the `esp_rmaker_report_node_details()` API after you add/remove the device. To see the logs, you have to set the Log Output level to "Debug" from `idf.py menuconfig`.

jacek12345, 2023-01-18:
Thank You @sanketwadekar 
I know and use `esp_rmaker_report_node_details()` API.
So can implement const[esp_rmaker_param_t](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html?highlight=valid_st#_CPPv418esp_rmaker_param_t) *param in RAM rather than FLASH and can modify this list before use  `esp_rmaker_report_node_details()`. It's clear now.
Next question is Can I change 'count' parameter after `esp_rmaker_param_add_valid_str_list(const[esp_rmaker_param_t](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html?highlight=valid_st#_CPPv418esp_rmaker_param_t) *param, const char *strs[], uint8_t count)`?
Or Can use `esp_rmaker_param_add_valid_str_list(const[esp_rmaker_param_t](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html?highlight=valid_st#_CPPv418esp_rmaker_param_t) *param, const char *strs[], uint8_t count)` a few times for the same parameter (or is there any API for re-create valid string list)?

sanketwadekar, 2023-01-18:
If you are increasing/decreasing items in the list of valid strings, then you should also change your `count` parameter accordingly. You can call the `esp_rmaker_param_add_valid_str_list` API multiple times even for the same parameter.

jacek12345, 2023-01-18:
thank You @sanketwadekar 
that's exactly what I wanted to get

jacek12345, 2023-01-18:
>You can call the esp_rmaker_param_add_valid_str_list API multiple times (with the updated list and count) for the same parameter.
I suggest adding this information to `esp_rmaker_param_add_valid_str_list` API description.

I succesfully implemented dynamic valid_str_list for parameter. Thank You.

jacek12345, 2023-02-24:
Can You confirm if calling  `esp_rmaker_param_update_and_report()` will report parameter valid_str_list also (earlier updated), or have to call `esp_rmaker_report_node_details()` ?

shahpiyushv, 2023-02-24:
Since valid_str_list is part of node config, you will have to call `esp_rmaker_report_node_details()` to update any changes related to it.

jacek12345, 2023-02-24:
Thank You @shahpiyushv 
Is it possible to add some API for valid_str_list report? We're using dynamic valid_str_lists and have to unnecessarily increase traffic on each updating list.

shahpiyushv, 2023-02-24:
The backend actually just overwrites whatever config is sent by the node. Adding a facility to update smaller snippets would require changes in backend logic including support for new topics for this. We may not consider this in near future. Are the changes really that frequent to cause an impact on network traffic?

jacek12345, 2023-02-24:
>Are the changes really that frequent to cause an impact on network traffic?
No, it is not raported very often. Usually while device configurating only. I though about AWS costs rather than network overload. 
`sometimes x thousands_of_users = bigger_AWS_costs`.
But I understand,  it doesn't make sense since it is not simple to add this API to backend and we can live with it.

closed 2023-02-24 by jacek12345

shahpiyushv, 2023-02-27:
@jacek12345 , AWS IoT messages are anyways metered in 5 KB increments. So, if your node config is less than 5KB, the optimisation of message size would not help. Moreover, ESP RainMaker now uses [basic ingest topics](https://docs.aws.amazon.com/iot/latest/developerguide/iot-basic-ingest.html) and so, there is no messaging cost, but only processing cost.

jacek12345, 2023-02-27:
Thank You for clarification.