espressif/esp-rainmaker #227: Can ESP RainMaker be used with BLE mesh?

Vanperdung, 2023-01-17:
Hi, I just found out ESP Rainmaker. Can ESP RainMaker be used with BLE Mesh? I will let the provisioner(gateway) connect to the ESP RainMaker cloud. Provisioner will process the MQTT messages to BLE Mesh mesages and vice versa. Is this possible?

Vanperdung, 2023-01-18:
Does anyone help me?

sanketwadekar, 2023-01-18:
Hi,
I am not sure whether this is possible. Is there a particular use case that you have identified for using BLE Mesh with Rainmaker?

sanketwadekar, 2023-01-19:
@Vanperdung 
 RainMaker uses only Wi-Fi after provisioning is done, and you can find a Wi-Fi mesh implementation [here](https://github.com/espressif/esp-mesh-lite/tree/master/examples/rainmaker_led_light). In your case, if you want to create a Bluetooth mesh, then you'll have to develop a local Bluetooth communication protocol that will translate to RainMaker devices/params.