espressif/esp-rainmaker #229: 'assert failed: spinlock_acquire....' during provisioning

jacek12345, 2023-01-19:
Everything works for some time. Since some moment i can't make provisioning. The only thing that fix it is erase-flash command. Then it works for some time.
I enabled Debug in menuconfig and see `assert failed: spinlock_acquire spinlock.h:123 ((result == SPINLOCK_FREE) == (lock->count == 0))`

Whole log:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1660
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (25) boot: compile time 15:16:06
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 8MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (64) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (72) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (79) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (87) boot:  4 ota_0            OTA app          00 10 00020000 003c0000
I (94) boot:  5 ota_1            OTA app          00 11 003e0000 003c0000
I (102) boot:  6 fctry            WiFi data        01 02 007a0000 00006000
I (109) boot:  7 devdata          WiFi data        01 02 007a6000 00003000
I (117) boot:  8 coredump         Unknown data     01 03 007a9000 00010000
I (124) boot: End of partition table
I (129) boot: No factory image, trying OTA 0
I (133) esp_image: segment 0: paddr=00020020 vaddr=3c0f0020 size=480ech (295148) map
I (195) esp_image: segment 1: paddr=00068114 vaddr=3fc9a240 size=04b24h ( 19236) load
I (199) esp_image: segment 2: paddr=0006cc40 vaddr=40374000 size=033d8h ( 13272) load
I (204) esp_image: segment 3: paddr=00070020 vaddr=42000020 size=ef288h (979592) map
I (385) esp_image: segment 4: paddr=0015f2b0 vaddr=403773d8 size=12e64h ( 77412) load
I (402) esp_image: segment 5: paddr=0017211c vaddr=50000000 size=00010h (    16) load
I (411) boot: Loaded app from partition at offset 0x20000
I (468) boot: Set actual ota_seq=1 in otadata[0]
I (468) boot: Disabling RNG early entropy source...
I (479) cpu_start: Pro cpu up.
I (479) cpu_start: Starting app cpu, entry point is 0x40375458
0x40375458: call_start_cpu1 at F:/GitHub/esp-idf-v4.4.3/components/esp_system/port/cpu_start.c:148

Fm(0) cpu_staDt:47p) epus :p.
_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (486) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 13 bit
D (493) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 24 bit
D (500) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (507) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (514) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (521) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 11 bit
D (528) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (535) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (542) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (549) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (556) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (563) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 13 bit
D (570) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 20 bit
D (577) efuse: In EFUSE_BLK1__DATA4_REG is used 5 bits starting with 27 bit
D (584) efuse: In EFUSE_BLK1__DATA5_REG is used 3 bits starting with 0 bit
D (591) efuse: In EFUSE_BLK1__DATA5_REG is used 8 bits starting with 3 bit
D (606) clk: RTC_SLOW_CLK calibration value: 3799194
I (615) cpu_start: Pro cpu start user code
I (615) cpu_start: cpu freq: 160000000
I (615) cpu_start: Application information:
I (618) cpu_start: Project name:     dtmiot1
[... 541 lines cut ...]
0x42026343: ble_hs_event_rx_data at F:/GitHub/esp-idf-v4.4.3/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:532

0x4202eaca: ble_npl_event_run at F:/GitHub/esp-idf-v4.4.3/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:125
 (inlined by) nimble_port_run at F:/GitHub/esp-idf-v4.4.3/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:78

0x42024706: nimble_host_task at F:/GitHub/esp-idf-v4.4.3/components/protocomm/src/transports/protocomm_nimble.c:475

0x403839fa: vPortTaskWrapper at F:/GitHub/esp-idf-v4.4.3/components/freertos/port/xtensa/port.c:131

ELF file SHA256: f6331a4a1df66736

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a20
0x40375a20: esp_restart_noos at F:/GitHub/esp-idf-v4.4.3/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1660
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
```

shahpiyushv, 2023-01-19:
Is this ust a normal Rainmaker project with your own params and the Provisioning timeout reduced to 2 min, with no other changes?

jacek12345, 2023-01-19:
Generally yes. + nimble for communication with sensors. No changes in RainMaker core. Only using available RainMaker API.

jacek12345, 2023-01-20:
without change #222 i have stack overflow when Debug enabled and this is a few rows earlier than last assert failed (that was with #222 added) so think that both are lack of memory issues. If disable Debug logging (free some memory) it works again (meaby on the "bleeding edge")
```
D (15095) protocomm_nimble: UUID did not match... 51
D (15095) protocomm_nimble: UUID (0x51) matched with proto-name = prov-session
D (15105) security1: Request to handle setup0_command
D (15115) gdma: new group (0) at 0x3fcc2634
D (15115) gdma: new pair (0,0) at 0x3fcc2670
D (15125) gdma: new
***ERROR*** A stack overflow in task ble has been detected.
```

closed 2023-01-20 by jacek12345

Tianbawen, 2024-03-08:
I also have this problem, how to solve it?

jacek12345, 2024-03-08:
Increase memory for nimble stack