espressif/esp-rainmaker #231: IDF v.5.0 - problem with ASYNCIO

jacek12345, 2023-01-24:
I installed idf v.5.0 and have this error:
`...\idf5.0_py3.7_env\Scripts\python.exe doesn't support asyncio. The issue can be worked around by re-running idf.py with the "--no-hints" argument.
`
Is other way than "--no-hints"?

I found that in v5. breaking changes:
`ASIO: Internal ASIO component and ASIO examples were migrated from IDF to https://github.com/espressif/esp-protocols/components/asio`
But still don't know what should do with it? (BTW: link is wrong)

sanketwadekar, 2023-01-31:
For IDF related issues, please file a new issue [here](https://github.com/espressif/esp-idf/issues)

jacek12345, 2023-02-07:
Thank You, resolved [(here)](https://github.com/espressif/esp-idf/issues/10712#issue-1572575717)

closed 2023-02-07 by jacek12345