espressif/esp-rainmaker #234: Rainmaker OTA problem

Assis08, 2023-02-01:
Hello everyone.

I'm using the following rainmaker example with ESP32-S3-WROOM-1-N16 and Arduino IDE:

 https://github.com/espressif/arduino-esp32/blob/master/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino

 together with the u8g2 library:

 https://github.com/olikraus/u8g2

after finishing rainmaker provisioning, the following error is occurring:

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (Tmr Svc) 
Core  0 register dump:
PC      : 0x40380bb3  PS      : 0x00060136  A0      : 0x80380d77  A1      : 0x3fcf6080  
A2      : 0x3fcc40bc  A3      : 0xffffffff  A4      : 0x00060f20  A5      : 0x3fcd3640  
A6      : 0xffffff01  A7      : 0x00000008  A8      : 0x80381cb0  A9      : 0x3fcf6090  
A10     : 0x00060123  A11     : 0x00000000  A12     : 0x00060120  A13     : 0x3fcd3640  
A14     : 0x00000000  A15     : 0xffffffe0  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x40380bb0:0x3fcf6080 0x40380d74:0x3fcf60c0 0x40379561:0x3fcf60e0 0x40379639:0x3fcf6110 0x42100025:0x3fcf6130 0x4210003f:0x3fcf6150 0x420d8ac3:0x3fcf6170 0x420d8afe:0x3fcf6190 0x420ce516:0x3fcf61b0 0x420cded1:0x3fcf61d0 0x420d3259:0x3fcf6210 0x420d35c9:0x3fcf6260 0x420cdd19:0x3fcf6280 0x420ce235:0x3fcf62c0 0x420c4a4a:0x3fcf6300 0x420c54b2:0x3fcf63d0 0x420c5d1d:0x3fcf6410 0x420c5d5e:0x3fcf6430 0x420bd3d2:0x3fcf6450 0x420bdfee:0x3fcf6470 0x42124cdf:0x3fcf6490 0x420addb8:0x3fcf64b0 0x420ade12:0x3fcf64d0 0x420af78d:0x3fcf64f0 0x420ad595:0x3fcf6530 0x4204d989:0x3fcf6550 0x4204e71e:0x3fcf6570 0x4204e747:0x3fcf66e0 0x40382e89:0x3fcf6700

when I disable the option: RMaker.enableOTA(OTA_USING_TOPICS); 
the error no longer happens.  Can anyone tell me if this could be a rainmaker bug with OTA?

sanketwadekar, 2023-02-02:
Hi @Assis08
Does the same error occur on the standalone RMakerSwitch example (without adding the u8g2 library) ?

Assis08, 2023-02-02:
Hi @sanketwadekar 

Yes, the problem occurs without the u8g2 library.

sanketwadekar, 2023-02-03:
@Assis08
Thanks for reporting this issue. The error seen is due to less stack memory allocated to "Tmr Svc" task. This can be configured using `CONFIG_FREERTOS_TIMER_TASK_STACK_DEPTH`. Increasing it's value will solve this issue. We will fix this in the next release of Arduino (2.0.7).