espressif/esp-rainmaker #237: Node ID maximum length

jacek12345, 2023-02-10:
What is maximum length of Node ID? Is it the only parameter that is unique for node?

shahpiyushv, 2023-02-10:
We use 3 different formats for node id based on the claiming and deployment types.

1. mac_address (12 bytes): 
2. uuid as base 64 encoded string (22 bytes)
3. uuid as standard text representation (36 bytes)

The cloud backend itself considers all 3 as text strings and so any other format would also work. The strings can even be longer (upto 64 bytes).

And yes, node_id is the only parameter unique for the node, but there is also a corresponding unique private key + certificate pair for each node and the node_id is embedded into the certificate.

jacek12345, 2023-02-10:
Thank You for response. What is the format after private deploy (no claiming)?
I need to read nodeid by `esp_rmaker_get_node_id()` and need to know size for storage array declaration.

shahpiyushv, 2023-02-10:
@jacek12345 , for private deployments, if you use the [rainmaker admin CLI](https://github.com/espressif/esp-rainmaker-admin-cli) the default is the 22 characters long base64 encoded UUID. It requires one additional byte for NULL termination.

However, note that in the private deployment, you can choose any format you want, but just need to ensure that the ids are unique. They could be mac addresses or even numbers like 000001, 0000002, so on.

closed 2023-02-18 by jacek12345