espressif/esp-rainmaker #241: Issue related schedules features

BharatNa, 2023-02-22:
### Is your feature request related to a problem?

Hi, 
i am currently working on a project related to agricultural controller using ESP32 wroom Module, 
 i need to turn on motor every day at a particular times like from 6 to 8 am like that but some day due to power cut some schedule not performing.. 

### Describe the solution you'd like.

Is there any way to give schedule should trigger the switch in bounded time  or when power came in it can recall the schedule which was not performed earlier...?

shahpiyushv, 2023-02-27:
Currently, there is no way for the firmware to know about missed schedules and for most cases, the exact time is important and any delay due to power off may not make sense. Eg. A schedule for turning on an ac/light at 7pm can't be triggered at 9am, just because the device was powered off at 7pm and came back at 9am.

Adding the support could be hard and even if we do it, it may not fit in the current framework, wherein the schedules infra just invokes the application registered callback and the callback executes the action without really knowing if this was a missed event or actual event. An option you have is to check if the [callback source](https://github.com/espressif/esp-rainmaker/blob/master/components/esp_rainmaker/include/esp_rmaker_core.h#L174) is `ESP_RMAKER_REQ_SRC_SCHEDULE` and if it is, capture the timestamp using below snippet and then store it in nvs

```
    time_t now;
    /* Get current time */
    time(&now);
```

On bootup, you can check this timestamp and compare with current time to see if any schedule may have been missed.

CC-MASTER05, 2023-03-01:
Hi,
You can add one RTC in your hardware and try to make one deep sleep until your schedule time come once RTC time reach at your schedule time at a time generate one interrupt and using those interrupt make you MCU alive add this logic in your code.