espressif/esp-rainmaker #242: Deleting device from node - clarification needed

jacek12345, 2023-02-24:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

There is no API to remove parameter from device, so I'm removing device and creating it one more time with desired parameters. 
Shuld I  use `esp_rmaker_node_remove_device()` and `esp_rmaker_device_delete()`  also, even when creating the same device for a while? I have not found in the documentation a detailed explanation of what these functions do underneath.

sanketwadekar, 2023-02-24:
@jacek12345
As the API name suggests, `esp_rmaker_node_remove_device()` removes the device from the node, and `esp_rmaker_device_delete()` frees all the memory resources allocated to the device, including device parameters and attributes. To remove a device, you should call `esp_rmaker_node_remove_device()` before `esp_rmaker_device_delete()`.

jacek12345, 2023-02-27:
Thank You @sanketwadekar

closed 2023-02-27 by jacek12345