espressif/esp-rainmaker #243: Claiming failed with error : Claim init failed

HassHasib, 2023-02-25:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

Arduino Ide 2.0.3

### Operating System used.

Windows

### How did you build your project?

Arduino IDE

### Development Kit.

Esp32 DevKit V1

### What is the expected behavior?

i want to connect my esp to the rainmaker app. in my new esp32.,

### What is the actual behavior?

I'm getting 
Claiming failed with error : Claim init failed
error in the app, after scanning the qr code. it never happened before.
Someone please help me.
[image]

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build Logs.

```plain
Reset Button Pressed!
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0

Starting ESP-RainMaker

Provisioning Started with name "PROV_TechStudyCeooalll" and PoP "TSC123RFV332" on BLE
Scan this QR code from the ESP RainMaker phone app.
                                          
  █▀▀▀▀▀���   █ ▄▄█▀▀▀▄▀  █▀▄▀▄█▀ █▀���▀▀▀█   
  █ ███ █ █▀█▄▄▄█▀���  ▀▄██  ▄█▄  █ ███ █   
  █ ▀▀��� █ █▄██▀▀▀▄▄██▀▀▀▀ ▄▄▄���█ █ ▀▀▀ █   
  ▀▀▀▀▀▀▀ █▄█▄��� ▀▄█ █ █▄▀▄▀ ▀▄▀ ▀▀▀▀▀▀��   
  █ ▀█▀█▀▄ ▄▀█▀▄▀▄▄▀█▄ ▄▀▄  ███▄██▀█▀▄    
   ▀▀  ▄��██▄▀▀    ▄▀▀ ▄█▀▄█ █▀▀▀▀ ��▀▀▀▀   
  █▀██▄ ▀█▄▀ ▀███ ��▀▄▄▄▄▀  ▀ ▀▄▄▀█▀ ▀ ▀   
  █▄ █ ▀▀▄▄▄▄█▀█▀█▄▀█ ██▄▄ ▀���▀▄▀█ ▀▀▀ █   
  █ ▄▀██▀▀ █��▀ ▀ ▄ ▀▄▄█▄▀██▀▄ ▀██ ▀█▀���    
  █▄▄█ ▄▀█▀▄▄▄ █▀██  ▄ ���█ ▀▀ ▀ ▄█ █▀ ▄▀   
  ▀ ▄ █▀▀ �� ▀▀  █ ▄▀▄▄█ ▀▀▀▀▄▀▄ ▀ ▀▀��     
  █▀█ █ ▀ █  ▀▀███▄▀█ ▄▄���▄█▀▄▀ ▄▀ █   ▀   
  ▄▄▄█▀ ▀▄���▀█▀▀▄▀ ██▄▄█▄█▄██▄▀▀▀���▀▀ █▄▀   
  █ █▀█ ▀▀ ▄  █▄█▀ ▀ ▄█▄▄▄█▀ █▀ ▀ ▀ ▀█▀   
  ▀ ��  ▀▀ █▄▄ ▀▀ ▀▄█▄██▄▀  ▀ ██���▀▀█▄▀ ▀   
  █▀▀▀▀▀█ ▄▄█    ���▀ ▄▄▀▄▄ █▀▄▄█ ▀ ██▀ █   
  �� ███ █ █▄▄    ▀ █▄▄▄ █▄█▀▄█���▀████▀█▄   
  █ ▀▀▀ █ ▀█  ▀ ���██▀   █ ▄█▀  █ ▄ ▄█ ▀█   
  ▀▀���▀▀▀▀ ▀ ▀ ▀▀ ▀ ▀  ▀▀▀   ▀▀ ▀   ▀▀▀▀   
                                          

If QR code is not visible, copy paste the below URL in a browser.
https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_TechStudyCeooalll","pop":"TSC123RFV332","transport":"ble"}
```

shahpiyushv, 2023-02-27:
Please update the Android app to the current latest (2.10.1) and retry.

FernandoGarcia, 2023-02-28:
Hi @shahpiyushv !

I have this version of APP installed and there's no way to configure my ESP32-Wrover IE with basic example.

I tried on tablet and smartphone.

When I managed to connect to board it says no network found, placing the SSID and password manually doesn't solve.

Almost all time the APP freezes in QR code scan.

Best regards.

shahpiyushv, 2023-02-28:
@FernandoGarcia since this sounds like a different issue, can you file it separately? The default issue form will also help us get more information about it. Please also mention what phone OS you are using while filing the new issue.

FernandoGarcia, 2023-02-28:
Done.

FernandoGarcia, 2023-03-01:
I found an error from my side while placing SSID and password manually.

I had  an extra space at the end of SSID added by auto complete from keyboard.

I have reported it [here](https://github.com/espressif/esp-rainmaker/issues/247).