espressif/esp-rainmaker #245: ESP rainmaker app says " No internet connection"

Jo14532, 2023-02-27:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

My ESP rainmaker app now continuously says "No internet connection" at the bottom of the screen. I can confirm I am connected to the internet and can watch youtube videos and use google.

Why do I only see this even though I can trigger my system using the on/off switch?

[image]

khushbushah2302, 2023-02-28:
Please update the Android app to the current latest (2.10.1) and retry.

Jo14532, 2023-02-28:
All fixed after the update thank you!

bxslp, 2023-03-06:
Where is the Android app(2.10.1)?  Would you please give one. Thank you very much!

khushbushah2302, 2023-03-06:
Hi @bxslp , 
ESP RainMaker Android app with version 2.10.1 is live on Google PlayStore. And code is available on GitHub latest release ([Release 2.10.1](https://github.com/espressif/esp-rainmaker-android/releases/tag/v2.10.1)).

closed 2023-03-06 by shahpiyushv

bxslp, 2023-03-06:
All fixed after the update. thanks !