espressif/esp-rainmaker #246: PWM not generate on PIN18 on ESP32-S3

CC-MASTER05, 2023-02-28:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I have tested led-light example by default on ESP32-S3 in pin 38 but now i switch this example on PIN 18 but didn't get any voltage on those pin also i have tested same way to switch example but got a same result as pin 38 working fine but once i change pin doesn't work.

can you help me to solve this and which which configuration i need to change?

IDF version : 4.4.4
Hardware: ESP32-S3-DevKitc-1

Log:
I (3060) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (3070) esp_rmaker_core: Waiting for MQTT connection
I (4100) esp_rmaker_time: SNTP Synchronised.
I (4100) esp_rmaker_time: The current time is: Tue Feb 28 12:04:53 2023 +0530[IST], DST: No.
I (4750) esp-x509-crt-bundle: Certificate validated
I (6590) esp_mqtt_glue: MQTT Connected
I (6590) esp_rmaker_node_config: Reporting Node Configuration of length 1903 bytes.
I (6610) esp_rmaker_param: Params MQTT Init done.
I (6610) esp_rmaker_param: Reporting params (init): {"Light":{"Name":"Light","Power":true,"Brightness":25,"Hue":180,"Saturation":100},"Time":{"TZ":"Asia/Kolkata","TZ-POSIX":"IST-5:30"},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"f72c1c12","Type":1}}
I (6630) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (6640) esp_rmaker_ota_using_topics: Subscribing to: node/68B6B33CD3D8/otaurl
I (11640) esp_rmaker_ota_using_topics: Fetching OTA details, if any.
I (12330) esp_rmaker_param: Received params: {"Light":{"Power":false}}
I (12330) app_main: Received write request via : Cloud
I (12330) app_main: Received value = false for Light - Power
I (12340) esp_rmaker_param: Reporting params: {"Light":{"Power":false}}
I (14180) esp_rmaker_param: Received params: {"Light":{"Power":true}}
I (14190) app_main: Received write request via : Cloud
I (14190) app_main: Received value = true for Light - Power
I (14190) esp_rmaker_param: Reporting params: {"Light":{"Power":true}}

as per log everything is okay but voltage not detected in PIN 18 as i got voltage on PIN 38 (default RGB light connected)

CC-MASTER05, 2023-03-01:
Hello Ranmakers,

can any buddy help me to figure out this issue?

CC-MASTER05, 2023-03-01:
@sanketwadekar  @shahpiyushv  can you help me?

shahpiyushv, 2023-03-01:
@CC-MASTER05 , I think it would be better to ask under [esp-idf issues](https://github.com/espressif/esp-idf/issues) to get relevant help from appropriate people who know more about the hardware and esp idf drivers.

CC-MASTER05, 2023-03-01:
actually this functionality working perfectly in esp-idf example but not working on esp-rainmaker example  @shahpiyushv

CC-MASTER05, 2023-03-02:
Can anyone  help me with rainmaker as i have done with esp-idf and it work fine i have issue with only rainmaker
@shahpiyushv

shahpiyushv, 2023-03-02:
@CC-MASTER05 the led support in RainMaker is just copied from [led_strip_rmt_ws2812.c](https://github.com/espressif/esp-idf/blob/release/v4.4/examples/common_components/led_strip/led_strip_rmt_ws2812.c) file of esp-idf and split into 2 files here:

[led_strip_rmt_ws2812.c](https://github.com/espressif/esp-rainmaker/blob/master/components/ws2812_led/led_strip_rmt_ws2812.c)

[ws2812_led.c](https://github.com/espressif/esp-rainmaker/blob/master/components/ws2812_led/ws2812_led.c)

Have you tried the [blink](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/get-started/blink) example from esp-idf which just uses this driver like how it is in esp-rainmaker?

CC-MASTER05, 2023-03-02:
Yes i have try with the blink example but my issue is that i am able to test every example with default pin and it work as per our expected.

**ISSUE:**
-> Change Pin  not getting output

**After Expected Result**
-> as before run with default pin same output need to observe after change pin

**After Get Result**
-> Not getting any output in changed pin

I have tested all the example but result is same in all

### INSHORT:
When i change pin output not getting in any example

CC-MASTER05, 2023-03-02:
Actually i thought when we change we missing something at build time that's why this happening if similar observe with you then we make sure this is our bug else if have any solution please give @shahpiyushv

shahpiyushv, 2023-03-02:
@CC-MASTER05 , it was not clear if the blink example of esp-idf worked with GPIO 18 for you or not.

CC-MASTER05, 2023-03-03:
yes it working

sanketwadekar, 2023-03-03:
Hi @CC-MASTER05 
I tried the switch example on s3 and gpio pin 18 with an actual led connected to it. Everything worked well. Can you tell me what variable/config you set to change the output pin?

CC-MASTER05, 2023-03-03:
@sanketwadekar  I have just changed pin by using idf.py menuconfig -> componentconfig -> WS2812 RGB LED

CC-MASTER05, 2023-03-03:
@sanketwadekar  can you write your test step so i can do same as do if i have miss something

CC-MASTER05, 2023-03-03:
```
`esp_err_t ws2812_led_init(void)
{
   rmt_config_t config = {
       .rmt_mode = RMT_MODE_TX,
       .channel = 0,
       .gpio_num = 18,
       .mem_block_num = 1,
       .tx_config.carrier_freq_hz = 38000,
       .tx_config.carrier_level = RMT_CARRIER_LEVEL_HIGH,
       .tx_config.idle_level = RMT_IDLE_LEVEL_LOW,
       .tx_config.carrier_duty_percent = 0,
       .tx_config.loop_en = 1,
       .tx_config.idle_output_en = 1,
       .tx_config.idle_level = 0,
   };
   // set counter clock to 40MHz
   config.clk_div = 2;

   if (rmt_config(&config) != ESP_OK)
   {
      ESP_LOGE(TAG, "RMT Config failed.");
      return ESP_FAIL;
   }
   if (rmt_driver_install(config.channel, 0, 0) != ESP_OK)
   {
      ESP_LOGE(TAG, "RMT Driver install failed.");
      return ESP_FAIL;
   }

   // install ws2812 driver
   led_strip_config_t strip_config = LED_STRIP_DEFAULT_CONFIG(1, (led_strip_dev_t)config.channel);
   g_strip = led_strip_new_rmt_ws2812(&strip_config);
   if (!g_strip)
   {
      ESP_LOGE(TAG, "Install WS2812 driver failed.");
      return ESP_FAIL;
   }
   return ESP_OK;
}
`
```

I have made some inside function and configure hardcore GPIO it working for me but not once i do off at a time completely off.
this is not a right solution but something as per result happening i want more specific result can you help me here to completly off.

sanketwadekar, 2023-03-03:
https://user-images.githubusercontent.com/67091512/222687607-b4137555-8ff0-463e-a839-6285f0414ec2.mp4

@CC-MASTER05 
As you can see in the video that I have recorded, the only config I have changed is the led gpio pin number `CONFIG_WS2812_LED_GPIO`.  I have attached pin-18 of esp32s3 to the "DI" pin of the ws2812 led strip. I have NOT changed any code in the driver/example.

CC-MASTER05, 2023-03-03:
Is there require any special environment set up as i have follow document to setup and i am able to execute so

sanketwadekar, 2023-03-09:
> Is there require any special environment set up as i have follow document to setup and i am able to execute so

No special environment is required.

CC-MASTER05, 2023-05-13:
Hello,

I have done this issue by one step down esp-idf version.

Thanks for the great support.

Thanks,
Fenil

closed 2023-05-13 by CC-MASTER05