espressif/esp-rainmaker #248: MQTT_CLIENT:  MQTT connect failed

Ph-tech06, 2023-03-04:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Failed to read from DHT sensor!
E (25417) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
E (25417) esp_mqtt_glue: MQTT_EVENT_ERROR
E (25417) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (25423) MQTT_CLIENT: MQTT connect failed
[image]
[image]

Ph-tech06, 2023-03-04:
When i upload code its work perfectly but after some days ( like 3-4 days) ... show many error ..
1. wifi is connected with esp32 but on rainmaker its show "offline"
2. esp32 connect with my router perfectly because my esp32 mac address show in my router page that show on ARP list but esp32 now working with alexa , google assitant.
3.if i re-upload the code then again work perfectly but after some days same error show again -------

E (25417) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=119
E (25417) esp_mqtt_glue: MQTT_EVENT_ERROR
E (25417) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1
E (25423) MQTT_CLIENT: MQTT connect failed

sanketwadekar, 2023-03-04:
@Ph-tech06 Please refer to the Rainmaker Fair Usage policy [here](https://www.esp32.com/viewtopic.php?f=41&t=32450&sid=68bc7eb051d9bd9a09121647f11a4030). Make sure your device is not sending too many messages to the cloud.
PS: Please use the "Runtime Bug report" form for creating issues that are related to crashes/errors as it helps us understand the issue much better.

Ph-tech06, 2023-03-04:
I have some doubt about messages sending via my device....
I don't understand this thing ...
Can I know how may i reduce the sending messages and how will I use fair
....
I mean which steps we are following to not face this type of error ..

sanketwadekar, 2023-03-04:
Your esp32 device uses MQTT to communicate with the Rainmaker cloud. If you are constantly updating Rainmaker device parameters and reporting it to the cloud using `updateAndReport()` API, you will exhaust the MQTT limit and the node will get blocked.

techiesms, 2023-03-12:
I think it's the issue of MQTT Budgeting. Well watch this video and you'll find it's solution in the same 
https://youtu.be/pNJBaWZ_vj0

shukur-alom, 2023-04-28:
when my wifi disconnect to esp32 esp rainmaker app goes offline it's ok. but when wifi turn on, esp32 never connect to wifi automatically and rainmaker app always show offline.
but when i press esp32 EN button it connect wifi prefectly and esp32 come to online. How to fix it?

shahpiyushv, 2023-04-28:
@shukur-alom , can you try the solution provided [here on the forums](https://www.esp32.com/viewtopic.php?f=41&t=29011)?

shukur-alom, 2023-04-29:
> @shukur-alom , can you try the solution provided [here on the forums](https://www.esp32.com/viewtopic.php?f=41&t=29011)?

when i switch esp32 2.0.0 to 2.0.6 it's working now.

shukur-alom, 2023-04-29:
@shahpiyushv 
when i switch esp32 2.0.0 to 2.0.6 it's working now.

rajsimiot, 2023-05-03:
I am facing the same issue . Even if I send data once per minute. Basically around 2000 messages in a day while the document says it gets blocked if we send 25000 msgs in a day

Ph-tech06, 2023-05-04:
> I am facing the same issue . Even if I send data once per minute. Basically around 2000 messages in a day while the document says it gets blocked if we send 25000 msgs in a day
/
/

Ph-tech06, 2023-05-04:
> I am facing the same issue . Even if I send data once per minute. Basically around 2000 messages in a day while the document says it gets blocked if we send 25000 msgs in a day

mail me - pritam.alexa@gmail.com
 share the code & Related Error

closed 2023-05-04 by Ph-tech06

GauravXs, 2024-08-20:
is the error solved?? I am facing the same error

djorr5, 2024-10-17:
Same issue here. I emailed you @Ph-tech06 . Screen shot of the error that occurs at random times:

[image]

shahpiyushv, 2024-10-18:
If MQTT disconnects and then reconnects at some later point of time, it is generally due to network issues (Internet and/or Wi-Fi) which may cause the MQTT keep alive packets to get missed.

djorr5, 2024-10-19:
@shahpiyushv I understand that network issues may cause it but I have seen 80+ disconnect and reconnect messages sometimes, and my wifi or network connection to the board was not affected during this time. Even in Insights I don't even see the RSSI drop below -55dB. 

In regards to the keep alive packets, where can I change the setting so that I can handle missed packets better. I.e are there settings in menuconfig to alter the timeouts? Retry messages etc?

It is annoying as these messages will swamp the Rainmaker app with offline alarm message then immediately with online alarm messages. Repeatedly as well without the ability to mute the alarm.