espressif/esp-rainmaker #249: esp rainmaker app failed to confirm node association

HassHasib, 2023-03-05:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm getting this error in app (esp rainmaker app failed to confirm node association)

And also in Arduino ide serial monitor i'm getting this error 
E (173618) esp-tls: [sock=48] select() timeout
E (173619) esp-tls: Failed to open new connection
E (173619) TRANSPORT_BASE: Failed to open a new connection
E (173621) MQTT_CLIENT: Error transport connect
E (173626) esp_mqtt_glue: MQTT_EVENT_ERROR

Please anyone tell me how can I fix this ?

HassHasib, 2023-03-08:
Someone reply please

sanketwadekar, 2023-03-08:
which version of ESP32-Arduino are you using? Also, have you updated the ESP Rainmaker app to the latest version?

techiesms, 2023-03-12:
Hello,

Recently I made a video that covers all the common errors people face while using ESP Rainmaker along with their solution. So do watch that video out and you may find the solution in that video 
https://youtu.be/pNJBaWZ_vj0

HassHasib, 2023-03-13:
Thanks a lot bro