espressif/esp-rainmaker #250: ESP32 RAINMAKER - "Failed To Allocate Memory"

coding-ram, 2023-03-09:
Hello , I am facing the error issue after uploading. Please check the screenshot below.  The Serial Monitor also gives a big error. Please help me out.......
[image]
[image]
As soon as I provide the Wifi credentials , the app shows the following error.
[image]

_Originally posted by @coding-ram in https://github.com/espressif/arduino-esp32/issues/7506#issuecomment-1461683535_

sanketwadekar, 2023-03-09:
Can you try using ESP32-Arduino 2.0.6  from the Boards manager?

Rudigao, 2023-03-11:
Hello.
I had exactly this same problem. ESP32 DEVKIT V1.
So I downgraded the esp32 boards to version 2.0.4 and it worked.

cane253, 2023-03-15:
Pls help how to downgrade it keeps updating to latest version. Even I have the same problem

Rudigao, 2023-03-17:
[image]

PrzemekWasiak, 2023-04-01:
I had the same problem already with version 2.0.6. After changing to 2.0.5, the same program works fine. Unfortunately, in 2.0.7 it still throws errors.

santoshurlam, 2024-07-01:
I have also same issue with Sending wifi credentials failed to associate device in esp RainMaker app with my ESP32 -Wroom 32D module , I'm using esp32 3.0.2 version board manager , I've tried by downgrading to 2.0.5,2.0.6 it is giving compilation error while uploading basic rainmaker example switch code.
Provide any solution to this issue .