espressif/esp-rainmaker #252: unable to compile after disabling the MQTT BUDGET

Sundareswar23, 2023-03-21:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.0.1-3-g886e98a2c1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32S3

### What is the expected behavior?

successful build with disabling the MQTT budget feature

### What is the actual behavior?

disable the feature try out the MQTT connection stability

### Steps to reproduce.

1. open any examples of esp-rainmaker
2. run idf.py menuconfig
3. under "ESP RainMaker Config" disable "Enable MQTT budgeting"
4. Run idf.py build

### Build Logs.

```plain
components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:143:1: error: unknown type name 'esp_err_t'
  143 | esp_err_t esp_rmaker_mqtt_budgeting_init(void)
      | ^~~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_init':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:149:12: error: 'ESP_FAIL' undeclared (first use in this function)
  149 |     return ESP_FAIL;
      |            ^~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:149:12: note: each undeclared identifier is reported only once for each function it appears in
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: At top level:
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:152:1: error: unknown type name 'esp_err_t'
  152 | esp_err_t esp_rmaker_mqtt_budgeting_deinit(void)
      | ^~~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_deinit':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:154:12: error: 'ESP_FAIL' undeclared (first use in this function)
  154 |     return ESP_FAIL;
      |            ^~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: At top level:
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:157:1: error: unknown type name 'esp_err_t'
  157 | esp_err_t esp_rmaker_mqtt_budgeting_stop(void)
      | ^~~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_stop':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:159:12: error: 'ESP_FAIL' undeclared (first use in this function)
  159 |     return ESP_FAIL;
      |            ^~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: At top level:
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:162:1: error: unknown type name 'esp_err_t'
  162 | esp_err_t esp_rmaker_mqtt_budgeting_start(void)
      | ^~~~~~~~~
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_start':
[... 39 lines cut ...]
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:155:1: error: control reaches end of non-void function [-Werror=return-type]
  155 | }
      | ^
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_stop':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:160:1: error: control reaches end of non-void function [-Werror=return-type]
  160 | }
      | ^
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_budgeting_start':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:165:1: error: control reaches end of non-void function [-Werror=return-type]
  165 | }
      | ^
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_increase_budget':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:170:1: error: control reaches end of non-void function [-Werror=return-type]
  170 | }
      | ^
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_decrease_budget':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:175:1: error: control reaches end of non-void function [-Werror=return-type]
  175 | }
      | ^
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c: In function 'esp_rmaker_mqtt_is_budget_available':
/Users/sundareswarpalanichamy/projects/ESP/esp-rainmaker/components/esp_rainmaker/src/mqtt/esp_rmaker_mqtt_budget.c:180:1: error: control reaches end of non-void function [-Werror=return-type]
  180 | }
      | ^
cc1: some warnings being treated as errors
```

shahpiyushv, 2023-03-21:
@Sundareswar23 , we will look into this, but please note that devices sending high number of messages will be deactivated as per our fair usage policy. What exactly is your use case which requires disabling budgeting?

Sundareswar23, 2023-03-21:
@shahpiyushv I am validating it, will let you know

Sundareswar23, 2023-03-22:
@shahpiyushv  i have customised the above application, publishing the messaging will be low. so i thought of disabling the mqtt budget and found this issue.

shahpiyushv, 2023-03-23:
@Sundareswar23 , the issues has been fixed now. Please check with the latest rainmaker agent code.

closed 2023-03-27 by Sundareswar23