espressif/esp-rainmaker #253: Esp rainmaker is not connecting to Google home

Gameboy134, 2023-03-23:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I was working on the project and in the morning it was working fine from the esprain maker app but was not connecting to my Google home I switched accounts but always the same error message shows up on the Google home works with Google page it say could not connect with esp rainmaker try again and still doesn't work after 24 hours of finding the problem

shahpiyushv, 2023-03-23:
@Gameboy134 , we have identified the issue and are working on a fix for it. It should be done in a day's time.

shahpiyushv, 2023-03-24:
@Gameboy134 , @luckyseptian17 This is fixed now. Can you check?

Gameboy134, 2023-03-24:
Thank you working well now very friendly customer support email

closed 2023-03-24 by shahpiyushv

laurentiuO, 2023-04-01:
Hello, I have the same problem. How can I fix it ? Thank You .

bk20dev, 2023-07-11:
Hey, I've recently run into the same problem. It turned out that I'd been setting up my device without all necessary params. After fixing my bug I flashed the code and suddenly I was able to add RainMaker to Google Home. As a hint, the same message also appeared when I sent some incorrect data through the param - like JSON instead of int.

Eidvargas96, 2023-09-11:
Hi, I am from Colombia.
I have not been able to connect the ESP Rainmaker account with Google Home.
It says: No compatible devices were found in your ESP RainMaker account.
help.
thank you.
[image]

shahpiyushv, 2023-09-11:
Hi @Eidvargas96 , what type of device have you added in your RainMaker account?

Eidvargas96, 2023-09-11:
I have the same problem with Alexa

shahpiyushv, 2023-09-11:
I just tried out the switch example and re-linked GVA and Alexa accounts and they worked well for me, so there does not seem any issue with the services in general. Can you double check if the account you are using from your RainMaker phone app and the ones in Alexa/GVA are same? Can you also share your registered email address and country from where you are trying this by sending an email to esp-rainmaker-admin@espressif.com?

Eidvargas96, 2023-09-11:
Could you try the dimmer example please.
thank you

pablojgauto, 2024-05-20:
Hello, I'm having the same issue. I can't associate with google home: "ESP RainMaker Could not be found. Try again".
[image]

shahpiyushv, 2024-05-20:
If there is no device compatible with GVA, it may throw such errors. What device types do you have in your account?

pablojgauto, 2024-05-20:
Thank you, I've defined a Swith and a Garage door. The devices work well in the Rainmaker app. 
Here is a exeprt of the code:

    /* Create a Switch device.
     * You can optionally use the helper API esp_rmaker_switch_device_create() to
     * avoid writing code for adding the name and power parameters.
     */
    switch_device = esp_rmaker_device_create(GARAGE_LIGHTS, ESP_RMAKER_DEVICE_SWITCH, NULL);
    porton_device = esp_rmaker_device_create(GARAGE_PORTON, ESP_RMAKER_DEVICE_GARAGE_DOOR, NULL);

    /* Add the write callback for the device. We aren't registering any read callback yet as
     * it is for future use.
     */
    esp_rmaker_device_add_cb(switch_device, write_cb, NULL);
    esp_rmaker_device_add_cb(porton_device, write_cb, NULL);

    /* Add the standard name parameter (type: esp.param.name), which allows setting a persistent,
     * user friendly custom name from the phone apps. All devices are recommended to have this
     * parameter.
     */
    esp_rmaker_device_add_param(switch_device, esp_rmaker_name_param_create(ESP_RMAKER_DEF_NAME_PARAM, GARAGE_LIGHTS));
    esp_rmaker_device_add_param(porton_device, esp_rmaker_name_param_create(ESP_RMAKER_DEF_NAME_PARAM, GARAGE_PORTON));

    /* Add the standard power parameter (type: esp.param.power), which adds a boolean param
     * with a toggle switch ui-type.
     */
    esp_rmaker_param_t *power_param_ch1 = esp_rmaker_power_param_create(ESP_RMAKER_DEF_POWER_NAME, DEFAULT_POWER);
    esp_rmaker_device_add_param(switch_device, power_param_ch1);
    
   /* Add a custom ACCIONAR parameter (type: NULL), which adds a boolean param
    * with a triger ui-type.
    */
    esp_rmaker_param_t *triger_param_ch2 = esp_rmaker_param_create("Accionar", NULL,
            esp_rmaker_bool(DEFAULT_POWER), PROP_FLAG_READ | PROP_FLAG_WRITE);
    if (triger_param_ch2) {
        esp_rmaker_param_add_ui_type(triger_param_ch2, ESP_RMAKER_UI_TRIGGER);
        esp_rmaker_device_add_param(porton_device, triger_param_ch2);
    } 

    /* Add a custom ESTADO parameter (type: NULL), which adds a boolean param
    * with a triger ui-type.
    */
    esp_rmaker_param_t *text_param_ch2 = esp_rmaker_param_create("ESTADO", NULL,
            esp_rmaker_str(port_ch1.t_position), PROP_FLAG_READ | PROP_FLAG_WRITE);
    if (text_param_ch2) {
        esp_rmaker_param_add_ui_type(text_param_ch2, ESP_RMAKER_UI_TEXT);
        esp_rmaker_device_add_param(porton_device, text_param_ch2);
    } 

    /* Assign the power parameter as the primary, so that it can be controlled from the
     * home screen of the phone apps. 
       NO LE AGREGAMOS AL PORTON PARA EVITAR APERTURAS EN FALSO
     */
    esp_rmaker_device_assign_primary_param(switch_device, power_param_ch1);

    /* Add this switch device to the node */
    esp_rmaker_node_add_device(node, switch_device);
    esp_rmaker_node_add_device(node, porton_device);

LordSizzz, 2024-12-30:
I'm having a similar issue, it seems like linking or relinking the rainmaker account after logging in and after the redirect back to google home, it just starts loading forever. Happened in 2 of my phones and I can't connect my esp32 nodes to google home again, I had it working a year ago, but I stopped working on the project and changed phones, the node connect and shows up in my rainmaker app list correctly, but linking the account to google home id stuck on a yellow circular loader forever.

shahpiyushv, 2024-12-30:
@LordSizzz , I just tried unlinking and re-linking my account and it worked fine for me, showing the below screen. What device types do you have in your account?
[image]

LordSizzz, 2024-12-30:
@shahpiyushv  I have a Light Matter node in my ESP32C6 board, I used the multiple examples, I can't manage to add the matter device directly to Google Home because I don't own a Matter Hub at home (that's what was showed up at least),but I can add my device to my rainmaker device list, I deleted my old ESP32 Light tho, so only my ESP32C6 with a Matter Light example is in the list.

shahpiyushv, 2024-12-30:
Matter is a local network protocol and so, the RainMaker app can control it only when it is in the same network as the device. Since Matter devices do not have cloud connectivity, they cannot be controlled remotely, either from RainMaker app or via Alexa/GVA. If the device was a pure RainMaker or a RainMaker + Matter device, it would have shown up in Alexa/GVA.