espressif/esp-rainmaker #254: unable to generate binary for gpio example

Sundareswar23, 2023-03-24:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v5.0.1-3-g886e98a2c1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32S3

### What is the expected behavior?

to generate correct gpio binary

### What is the actual behavior?

build fails in esp_rmaker_console.c

### Steps to reproduce.

1. git pull latest change in rainmaker repo
2. build the gpio example
3. error occurs in compilation

### Build Logs.

```plain
ichamy/esp/old_esp/esp-idf/components/esp_http_server/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/http_parser -I/Users/sundareswarpalanichamy/projects/ccd-pro/managed_components/espressif__mdns/include -I/Users/sundareswarpalanichamy/projects/ccd-pro/managed_components/espressif__esp_schedule/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/efuse/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/efuse/esp32s3/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/driver/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/driver/deprecated -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/driver/esp32s3/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/esp_pm/include -I/Users/sundareswarpalanichamy/esp/old_esp/esp-idf/components/esp_ringbuf/include -mlongcalls  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/Users/sundareswarpalanichamy/projects/ccd-pro=. -fmacro-prefix-map=/Users/sundareswarpalanichamy/esp/old_esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.1-3-g886e98a2c1\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/console/esp_rmaker_console.c.obj -MF esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/console/esp_rmaker_console.c.obj.d -o esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/console/esp_rmaker_console.c.obj -c /Users/sundareswarpalanichamy/projects/ccd-pro/managed_components/espressif__esp_rainmaker/src/console/esp_rmaker_console.c
/Users/sundareswarpalanichamy/projects/ccd-pro/managed_components/espressif__esp_rainmaker/src/console/esp_rmaker_console.c:16:10: fatal error: esp_rmaker_common_console.h: No such file or directory
   16 | #include <esp_rmaker_common_console.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[895/913] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/wifi.c.objninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /Users/sundareswarpalanichamy/projects/ccd-pro/build/log/idf_py_stderr_output_89215 and /Users/sundareswarpalanichamy/projects/ccd-pro/build/log/idf_py_stdout_output_89215
```

shahpiyushv, 2023-03-24:
Seems you may not have pulled the submodules using `git submodule update --init --recursive` after pulling the latest code. Can you check once?

Sundareswar23, 2023-03-24:
ok @shahpiyushv  , what if i have added same package in idf_component.yml file the build is getting failed.

below is the component.yml file, provide us suggestion to update below file

```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_rainmaker:
    path: components/esp_rainmaker
    git: https://github.com/espressif/esp-rainmaker.git
  espressif/esp_schedule:
    path: components/esp_schedule
    git: https://github.com/espressif/esp-rainmaker.git
  espressif/json_generator: ^1
  espressif/json_parser: =1.0.0
  espressif/qrcode:
    version: ^0.1.0
    rules:
      - if: "idf_version >=5.0"
  espressif/rmaker_common: ^1
```

shahpiyushv, 2023-03-24:
esp-rainmaker itself is not available via package manager yet, but you can pull in other components as required via package manager. Can you elaborate what the exact issue is, for which you need these components via package manager?

Sundareswar23, 2023-03-24:
@shahpiyushv  I am using the rmaker_mqtt api's for my application to connect with our cloud. 

till this commit https://github.com/espressif/esp-rainmaker/commit/c1b688511380008e98129b87b932cbb4c299257e
rainmaker compiled correctly. with latest build it is getting failed.

shahpiyushv, 2023-03-24:
@Sundareswar23 , that is not really an intended use case of the RainMaker's MQTT APIs. I would recommend using the underlying esp-mqtt component directly, which you can check out [here](https://github.com/espressif/esp-mqtt) which is part of esp-idf itself. If you still want to continue using esp-rainmaker, you may have to use it the same way as it is used in the standard RainMaker examples, which is without the package manager. 

Meanwhile, we have also updated the rmaker_common component to 1.3.0 in component manager, just in case it helps you resolve your issues.

closed 2023-03-24 by Sundareswar23