espressif/esp-rainmaker #256: rainmaker project is getting crashed when i do switching on / off fast -fast

sharmajiji, 2023-03-26:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Esp rainmaker project is getting crashed when i do switching on / off fast -fast. I used 4 channel relay module with esp32 and taking 4 input with internal pullup resistor of esp32 to control devices manually. but its getting crashed when i do switching fast. It starts giving norrmal data to garbage data on serial monitor. now  i have restart the board to work properly but again on switch it get crashed.

sharmajiji, 2023-03-28:
> @sharmajiji Can you share the serial terminal logs?

On serial Monitor
I am getting normally Serial.print command responses and suddenly ESP32 Stop working and start giving garbage data (for example like photo) https://user-images.githubusercontent.com/77848540/228333175-4c3458a3-41b1-42e5-9b36-2da6abcd99c3.jpg)

Its happen when I do switching fast manually or by app, Google home
It stops working from normal condition and start giving garbage data.
Then I have to press reset button to run again.

And Extra 
I am Taking 4 input in esp32 with internal pull-up resistor ,4 relays with esp32 fitted inside the switch Panelbox.

sharmajiji, 2023-03-28:
> @sharmajiji Can you share the serial terminal logs?

In addition i Joined a wiring sketch (https://user-images.githubusercontent.com/77848540/228338493-ce59f811-a5de-49d2-ac8d-d4002c9cf10b.jpg)

Sorry for bad drawing
Here is esp32 connect with normal switches and Relay 

Relay denoted by R, Switch denoted by S, A is Ac to Dc Adapter.

I also want to add that in one relay in 4 channel relay module is connected with manual regulator so that I can on off switch by App & switch and control speed only by manually from regulator

 i added a connection diagram of that (https://user-images.githubusercontent.com/77848540/228339484-00e576b0-a787-4c16-a04a-9cbc4d5e6194.jpg) 

but when I change the speed of fan then also esp32 get crashed.

sanketwadekar, 2023-03-29:
@sharmajiji
The information provided isn't sufficient to provide an answer. Can you share the logs on Serial Monitor and your sketch code? Also, are you switching using the physical buttons or the phone app?

sharmajiji, 2023-03-29:
> the

 using both manual home switches and phone app.

sharmajiji, 2023-03-29:
> @sharmajiji Can you share the serial terminal logs?

Serial Monitor (https://user-images.githubusercontent.com/77848540/228645050-2ccb8c24-0844-4f27-84d5-1b2d894ce3cb.jpg)

sharmajiji, 2023-03-29:
> @sharmajiji
> The information provided isn't sufficient to provide an answer. Can you share the logs on Serial Monitor and your sketch code? Also, are you switching using the physical buttons or the phone app?

serial monitor (https://user-images.githubusercontent.com/77848540/228645434-a774219b-bad7-4140-bbb7-1e8306143d14.jpg)

jacek12345, 2023-03-31:
Do You have flyback diodes on Your relays?

sanketwadekar, 2023-03-31:
> > Do You have flyback diodes on Your relays?
> 
> Yes,
> I am using 4 channel relay module low level trigger.
> I attached a photo of my relay module.
> (https://user-images.githubusercontent.com/77848540/229140892-0649e35a-d6b3-47e2-8504-14172bb33125.jpg)
> 
>
@sharmajiji 
Try removing all connections and then run the code on your device. If the issue persists, there might be issues in your code itself.

jacek12345, 2023-04-11:
What is your power source? Seems to be not enough current.

sharmajiji, 2023-04-11:
> What is your power source? Seems to be not enough current.

5v 2amp connected with ESP32 and parllel connected with my relay board

jacek12345, 2023-04-11:
As this is evidently hardware issue and it is not right place for resolving, I suggest to look closer on scheme and flyback diodes and investigate if there is maybe one canal that makes problem.