espressif/esp-rainmaker #258: Compilation fails because "esp_rmaker_factory_get_size" is not defined

me-no-dev, 2023-04-04:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP32-Arduino Lib Builder

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

Any

### What is the expected behavior?

Expected build to not fail

### What is the actual behavior?

Build fails: https://github.com/espressif/esp32-arduino-lib-builder/actions/runs/4607493703/jobs/8142082876#step:4:62180

### Steps to reproduce.

Just compile any project with RainMaker master (ESP-S3-BOX also failed to build)

### Build Logs.

```plain
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_client_data.c: In function 'esp_rmaker_get_client_cert_len':
62179
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_client_data.c:78:12: error: implicit declaration of function 'esp_rmaker_factory_get_size'; did you mean 'esp_rmaker_factory_get'? [-Werror=implicit-function-declaration]
62180
     return esp_rmaker_factory_get_size(ESP_RMAKER_CLIENT_CERT_NVS_KEY);
62181
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
62182
            esp_rmaker_factory_get
62183
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_client_data.c: In function 'esp_rmaker_get_mqtt_conn_params':
62184
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_client_data.c:128:27: error: 'esp_rmaker_mqtt_conn_params_t' {aka 'struct <anonymous>'} has no member named 'client_key_len'; did you mean 'client_key'?
62185
         mqtt_conn_params->client_key_len = esp_rmaker_get_client_key_len();
62186
                           ^~~~~~~~~~~~~~
62187
                           client_key
62188
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp_rainmaker/src/core/esp_rmaker_client_data.c:133:27: error: 'esp_rmaker_mqtt_conn_params_t' {aka 'struct <anonymous>'} has no member named 'client_cert_len'; did you mean 'client_cert'?
62189
         mqtt_conn_params->client_cert_len = esp_rmaker_get_client_cert_len();
62190
                           ^~~~~~~~~~~~~~~
62191
                           client_cert
62192
cc1: some warnings being treated as errors
62193
ninja: build stopped: subcommand failed.
```
```

Oopas, 2023-04-07:
I have also encountered this problem. Do you have any ideas?

shahpiyushv, 2023-04-07:
Have you pulled in the submodules too, using `git submodule update --init --recursive`?

me-no-dev, 2023-04-07:
@shahpiyushv we have definitely pulled all submodules. `esp_rmaker_factory_get_size` is nowhere to be found, but since `client_cert_len` is also undefined, I opted to remove lines 128 and 133 and also remove the whole `esp_rmaker_get_client_cert_len` function. That produced a successful build and everything seems to work too. I guess this is some old or unfinished feature that is no longer needed? Please have this resolved, because it's really blocking Arduino 2.0.8 (which was also initially blocked by RainMaker for unrelated malloc issue that is now fixed).

shahpiyushv, 2023-04-09:
One reason could be the order of including extra components. RainMaker examples have this

set(EXTRA_COMPONENT_DIRS ${RMAKER_PATH}/components/esp-insights/components ${RMAKER_PATH}/components ${RMAKER_PATH}/examples/common)

If the esp-insights is included AFTER RainMaker components, then rmaker_common from within esp-insights would get used , instead of the one one in esp-rainmaker, and that itself may not be up to date. This order will have to be checked in the arduino lib builder code. 

If Arduino lib builder uses idf component manager, such issues will get fixed soon, but this is blocked on some issues in idf component manager itself.

me-no-dev, 2023-04-09:
> If Arduino lib builder uses idf component manager, such issues will get fixed soon, but this is blocked on some issues in idf component manager itself.

it's starting to use it in the upcoming v3.0.0 based on IDF 5.1, but on 2.0.x it's not and will not (probably releasing the last such version). Will try the reorder and let you know. As a side note, couldn't this be done in a way that is not so touchy. I imagine any project that has not yet started using the component manager will suffer. Maybe this case can be checked and a more meaningful error message be produced?

me-no-dev, 2023-04-11:
@shahpiyushv adjusting the order of components helped! Thanks