espressif/esp-rainmaker #260: Questions about factory reset

Vieufoux, 2023-04-12:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I have some questions about factory reset:
- Will the aws iot certificate which stored in nvs be cleared after factory reset?
- If the certificates are cleared, if the same node invoke the claiming API again, is the content of the aws iot certificate generated by rainmaker cloud each time the same or different?
- The rmaker core will trigger a user mapping reset (if mqtt connection is active) so that the node gets removed from the user's account immediately. If the node is factory reset when mqtt connection is inactive, and is not connected to the Internet again, will the node always keep existing under the user's account?
- Can i still see the node on Rainmaker dashboard or ESP Insights after node factory reset? or will it be removed from dashboard after factory reset?

shahpiyushv, 2023-04-12:
> Will the aws iot certificate which stored in nvs be cleared after factory reset?

No, since that is stored in the fctry partition, which is separate from nvs.

> If the certificates are cleared, if the same node invoke the claiming API again, is the content of the aws iot certificate generated by rainmaker cloud each time the same or different?

Certificates will be cleared only if you explicitly erase the entire flash or the fctry partition using esptool or its variants (via Arduino as an example). Assuming that you cleared the certificates in this way, even then, for self claiming, the same node id will be assigned, but the certificate will be different.

> The rmaker core will trigger a user mapping reset (if mqtt connection is active) so that the node gets removed from the user's account immediately. If the node is factory reset when mqtt connection [...]

Yes, because the cloud has no way to know that the node was removed. However, if another user provisions that same node, it will then get removed from the older user's account.

> Can i still see the node on Rainmaker dashboard or ESP Insights after node factory reset? or will it be removed from dashboard after factory reset?

Since factory reset just erases the nvs partition and not the certificate information, the node will keep showing on RainMaker/Insights dashboard. Even if the flash itself is erased, the node will keep showing on dashboard. The node will disappear only if it is using self claiming and was added by some other user later.

Vieufoux, 2023-04-19:
@shahpiyushv When I was reading the Rainmaker document, I saw this note
[image]

So I have some other questions:
- When the client invokes the API to update node param, cloud will send MQTT message to node. Does the cloud respond to the HTTP client immediately after sending the MQTT message, or wait until the execution result of the node is received before responding?
- Will the cloud store Node Configuration and the value of parameters? When the client invokes the API to get the value of node param, whether the value of param is obtained directly from the cloud database or from node in real-time？

shahpiyushv, 2023-07-12:
Closing this issue since all questions have been answered.

closed 2023-07-12 by shahpiyushv