espressif/esp-rainmaker #261: The connection of the device is lost when the mobile APP is connected

perseverance51, 2023-04-21:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF v4.4.4-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP-IDF4.4

### What is the expected behavior?

The APP searches for the ESPThe sample project of esp-rainmaker\examples\switch compiled by espidf 4.4 was uploaded to the esp32 devkit v1 device. After searching for the Bluetooth device of ESP32, the mobile APP clicked Connect, but the device disappeared immediately.32 Bluetooth device, clicks Connect, and the device disappears again.

### What is the actual behavior?

The APP searches for the ESP32 Bluetooth device, clicks Connect, and the device disappears again.

### Steps to reproduce.

When you search for an esp32 Bluetooth device, click on the connected device and it disappears。
[image]

[image]

### Debug Logs.

```plain
[13:13:37.523] [0;32mI (1952) app_wifi: If QR code is not visible, copy paste the below URL in a browser.
[13:13:37.523] https://rainmaker.espressif.com/qrcode.html?data={"ver":"v1","name":"PROV_518642","pop":"ed768537","transport":"ble"}[0m
[13:13:37.523] [0;32mI (1962) app_main: Provisioning QR : {"ver":"v1","name":"PROV_518642","pop":"ed768537","transport":"ble"}[0m
[13:13:37.523] [0;32mI (1972) app_wifi: Provisioning Started. Name : PROV_518642, POP : ed768537[0m
[13:13:37.523] [0;32mI (1982) app_wifi: Provisioning will auto stop after 30 minute(s).[0m
[13:13:37.523] 
[13:13:51.060] [0;32mI (15542) NimBLE: GAP procedure initiated: advertise; [0m
[13:13:51.082] [0;32mI (15542) NimBLE: disc_mode=2[0m
[13:13:51.082] [0;32mI (15542) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256[0m
[13:13:51.082] [0;32mI (15552) NimBLE: 
[13:13:51.082] [0m
[13:13:51.082] 
[13:13:58.518] [0;32mI (23002) NimBLE: GAP procedure initiated: advertise; [0m
[13:13:58.541] [0;32mI (23002) NimBLE: disc_mode=2[0m
[13:13:58.541] [0;32mI (23002) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=256 adv_itvl_max=256[0m
[13:13:58.541] [0;32mI (23002) NimBLE: 
[13:13:58.541] [0m
[13:13:58.541]
```

khushbushah2302, 2023-04-26:
Hi @perseverance51 , 
What is the android app version ?
Is it possible to share android app logs ?

perseverance51, 2023-04-27:
> Hi @perseverance51 , What is the android app version ? Is it possible to share android app logs ?

I tested all of the following apps: ESP_RainMaker_2.10.1, 2.8.0, and ESP_BLE_Prov_2.1.0. I also tested the ESP RainMaker routine burning with the Arduino IDE and connected to the mobile APP.

shukur-alom, 2023-04-28:
when my wifi disconnect to esp32 esp rainmaker app goes offline it's ok. but when wifi turn on, esp32 never connect to wifi automatically and rainmaker app always show offline.
but when i press esp32 EN button it connect wifi prefectly and esp32 come to online. How to fix it?

shahpiyushv, 2023-04-28:
@shukur-alom , your query has been answered [here](https://github.com/espressif/esp-rainmaker/issues/248#issuecomment-1527082499). Just as an additional note, please avoid asking questions on issues unrelated to yours. This specific issue is about provisioning itself failing that too with esp-idf. Your issue is about device not reconnecting after losing Wi-Fi, that too (most probably) with Arduino.

shahpiyushv, 2023-04-28:
@perseverance51 , can you enable additional debug logs using idf.py menuconfig and retry?

Component Config --> Log Output --> Default Log Verbosity --> Debug

Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs 

Meanwhile, what is your phone model and Android version? Have you tried with a different phone?

perseverance51, 2023-04-30:
@shahpiyushv  I can connect to esp32 using other Bluetooth debugging apps. When I disconnected the ESP32 ble device and went back to the ESP RainMaker APP interface to try to connect to esp32, I was able to pull up the connection and successfully connect to the device. I'm not sure if it's because the third-party Bluetooth APP debug assistant successfully logged in or something.
[image]

perseverance51, 2023-04-30:
The same thing happens when you add a new esp32 Bluetooth device, so as long as the connected device has been successfully connected before and then re-paired, it won't be disconnected again.
- The following is the log printed when a new device is connected and disconnected:
`
[17:42:35.248] D (185578) protocomm_nimble: Inside BLE connect w/ conn_id - 1[0m
[17:42:35.248] D (185578) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0016 len=2
[17:42:35.248] [0m
[17:42:35.248] D (185578) NimBLE: 0x16 [0m
[17:42:35.248] D (185578) NimBLE: 0x20 [0m
[17:42:35.248] D (185588) NimBLE: 0x02 [0m
[17:42:35.248] D (185588) NimBLE: 0x01 [0m
[17:42:35.248] D (185588) NimBLE: 0x00 [0m
[17:42:35.248] D (185588) NimBLE: 
[17:42:35.248] [0m
[17:42:35.248] 
[17:42:37.093] D (187398) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0020 len=14
[17:42:37.093] [0m
[17:42:37.093] D (187398) NimBLE: 0x20 [0m
[17:42:37.093] D (187398) NimBLE: 0x20 [0m
[17:42:37.093] D (187398) NimBLE: 0x0e [0m
[17:42:37.093] D (187398) NimBLE: 0x01 [0m
[17:42:37.093] D (187408) NimBLE: 0x00 [0m
[17:42:37.093] D (187408) NimBLE: 0x06 [0m
[17:42:37.093] D (187408) NimBLE: 0x00 [0m
[17:42:37.093] D (187408) NimBLE: 0x06 [0m
[17:42:37.093] D (187418) NimBLE: 0x00 [0m
[17:42:37.093] D (187418) NimBLE: 0x00 [0m
[17:42:37.093] D (187418) NimBLE: 0x00 [0m
[17:42:37.093] D (187418) NimBLE: 0xf4 [0m
[17:42:37.093] D (187428) NimBLE: 0x01 [0m
[17:42:37.093] D (187428) NimBLE: 0x00 [0m
[17:42:37.093] D (187428) NimBLE: 0x00 [0m
[17:42:37.093] D (187428) NimBLE: 0x00 [0m
[17:42:37.093] D (187438) NimBLE: 0x00 [0m
[17:42:37.093] D (187438) NimBLE: 
[17:42:37.093] [0m
[17:42:37.093] D (187668) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=11 data=[0m
[17:42:37.438] D (187668) NimBLE: 0x07 [0m
[17:42:37.438] D (187668) NimBLE: 0x00 [0m
[17:42:37.438] D (187668) NimBLE: 0x04 [0m
[17:42:37.438] D (187668) NimBLE: 0x00 [0m
[17:42:37.438] D (187678) NimBLE: 0x10 [0m
[17:42:37.438] D (187678) NimBLE: 0x01 [0m
[17:42:37.438] D (187678) NimBLE: 0x00 [0m
[17:42:37.438] D (187678) NimBLE: 0xff [0m
[17:42:37.438] D (187688) NimBLE: 0xff [0m
[17:42:37.438] D (187688) NimBLE: 0x00 [0m
[17:42:37.438] D (187688) NimBLE: 0x28 [0m
[17:42:37.438] D (187688) NimBLE: 
[17:42:37.438] [0m
[17:42:37.438] D (187698) NimBLE: host tx hci data; handle=1 length=26
[17:42:37.438] [0m
[17:42:37.438] D (187698) NimBLE: ble_hs_hci_acl_tx(): [0m
[17:42:37.438] D (187708) NimBLE: 0x01 [0m
[17:42:37.438] D (187708) NimBLE: 0x00 [0m
[17:42:37.438] D (187708) NimBLE: 0x1a [0m
[17:42:37.438] D (187708) NimBLE: 0x00 [0m
[17:42:37.438] D (187718) NimBLE: 0x16 [0m
[17:42:37.438] D (187718) NimBLE: 0x00 [0m
[17:42:37.438] D (187718) NimBLE: 0x04 [0m
[17:42:37.438] D (187718) NimBLE: 0x00 [0m
[17:42:37.438] D (187728) NimBLE: 0x11 [0m
[17:42:37.438] D (187728) NimBLE: 0x14 [0m
[17:42:37.438] D (187728) NimBLE: 0x01 [0m
[17:42:37.438] D (187728) NimBLE: 0x00 [0m
[17:42:37.438] D (187738) NimBLE: 0xff [0m
[17:42:37.438] D (187738) NimBLE: 0xff [0m
[17:42:37.438] D (187738) NimBLE: 0xb4 [0m
[17:42:37.438] D (187748) NimBLE: 0xdf [0m
[17:42:37.438] D (187748) NimBLE: 0x5a [0m
[17:42:37.438] D (187748) NimBLE: 0x1c [0m
[... 109 lines cut ...]
[17:42:47.624] D (197908) NimBLE: 0x00 [0m
[17:42:47.624] D (197908) NimBLE: 0x00 [0m
[17:42:47.624] D (197918) NimBLE: 0x00 [0m
[17:42:47.624] D (197918) NimBLE: 0x00 [0m
[17:42:47.624] D (197918) NimBLE: 0x00 [0m
[17:42:47.624] D (197918) NimBLE: 0x00 [0m
[17:42:47.624] D (197928) NimBLE: 0x00 [0m
[17:42:47.624] D (197928) NimBLE: 0x00 [0m
[17:42:47.624] D (197928) NimBLE: 0x00 [0m
[17:42:47.624] D (197928) NimBLE: 0x07 [0m
[17:42:47.624] D (197938) NimBLE: 0x00 [0m
[17:42:47.624] D (197938) NimBLE: 
[17:42:47.624] [0m
[17:42:47.624] D (197938) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
[17:42:47.624] [0m
[17:42:47.624] D (197948) NimBLE: 0x0a [0m
[17:42:47.624] D (197948) NimBLE: 0x20 [0m
[17:42:47.624] D (197948) NimBLE: 0x01 [0m
[17:42:47.624] D (197958) NimBLE: 0x01 [0m
[17:42:47.624] D (197958) NimBLE: 
[17:42:47.624] [0m
[17:42:47.624] D (197958) protocomm_nimble: Minimum free heap size = 125188, free Heap size = 136116[0m
[17:42:47.624] 

`

perseverance51, 2023-04-30:
Below is the phone and Bluetooth information used

[image]

[image]

shahpiyushv, 2023-05-09:
@perseverance51 , we have never seen this issue and can't even reproduce and so, it is hard to debug. Have you tried changing either the phone or the board?