espressif/esp-rainmaker #262: WiFi event block other processes

nravin5, 2023-04-30:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

/* Wait for Wi-Fi connection */
    xEventGroupWaitBits(wifi_event_group, WIFI_CONNECTED_EVENT, false, true, portMAX_DELAY);

this function in app_wifi.c    blocks  other operations when there is no wifi....or not provisioning.

in case of we read push button input we can not get value until wifi get connected.

shahpiyushv, 2023-05-02:
This particular call just blocks the main task, and not any others. How exactly are you managing your button events?

sanketwadekar, 2023-05-05:
@nravin5 You should create a separate task to read the pin status.

shahpiyushv, 2023-05-05:
@nravin5, any reason why you aren't using interrupts? Does your application actually need some polling? Please remember to add a delay in your loop, even if you create your own task, so that it doesn't hog the CPU.

nravin5, 2023-05-07:
> @nravin5, any reason why you aren't using interrupts? Does your application actually need some polling? Please remember to add a delay in your loop, even if you create your own task, so that it doesn't hog the CPU.

At hardware level if there is any type wire connection issue or push pin rusting or any thing which creates random trigger  in that case the interrupts are disturbing the whole system.
In another case we can control when and how I want to read the pin values....
as my priority is run each function one by one not interrupting to others is my use case "home automation Project"

closed 2023-05-10 by nravin5