espressif/esp-rainmaker #265: When add two fan, the pharm speed value received when changing through app always from first fan if we change speed of second fan

nravin5, 2023-05-14:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### IDF / ESP32-Arduino version.

ESP-IDF Release v5.0.2 or ESP-IDF Release v5.0.1

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### Development Kit.

esp32c3

### What is the expected behavior?

working through visual studio 

my code ....
    /* Create a device and add the relevant parameters to it */
    fan_device = esp_rmaker_fan_device_create("Fan1", NULL, DEFAULT_POWER);
    esp_rmaker_device_add_cb(fan_device, write_cb, NULL);
    esp_rmaker_device_add_param(fan_device, esp_rmaker_speed_param_create(ESP_RMAKER_DEF_SPEED_NAME, DEFAULT_SPEED));
    esp_rmaker_node_add_device(node, fan_device);

    fan_device1 = esp_rmaker_fan_device_create("Fan2", NULL, DEFAULT_POWER);
    esp_rmaker_device_add_cb(fan_device1, write_cb, NULL);
    esp_rmaker_device_add_param(fan_device1, esp_rmaker_speed_param_create(ESP_RMAKER_DEF_SPEED_NAME, DEFAULT_SPEED));
    esp_rmaker_node_add_device(node, fan_device1);

### What is the actual behavior?

Only problem when speed changed throuch App

but with alexa app or voice works well.

may be issue with app or Rainmaker code. 

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
11:22:18.724 -> [0;32mI (752186) esp_rmaker_param: Received params: {"Fan1":{"Speed":2}}[0m
11:22:18.771 -> [0;32mI (752186) app_main: Received write request via : Cloud[0m
11:22:18.771 -> [0;32mI (752186) app_main: Received value = 2 for Fan1 - Speed[0m
11:22:18.771 -> [0;32mI (752196) esp_rmaker_param: Reporting params: {"Fan1":{"Speed":2}}[0m
11:22:19.897 -> [0;32mI (753316) esp_rmaker_param: Received params: {"Fan1":{"Speed":2}}[0m
11:22:19.897 -> [0;32mI (753316) app_main: Received write requesin: Received value = 2 for Fan1 - Speed[0m
11:22:19.897 -> [0;32mI (753316) esp_rmaker_param: Reporting params: {"Fan1":{"Speed":2}}[0m
11:22:20.084 -> [0;33mW (753536) protocomm_httpd: Closing session with ID: 1890314498[0m
11:22:26.561 -> [0;32mI (759976) esp_rmaker_param: Received params: {"Fan1":{"Speed":2}}[0m
11:22:26.561 -> [0;32mI (759976) app_main: Received write request via : Local[0m
11:22:26.561 -> [0;32mI (759976) app_main: Received value = ) esp_rmaker_param: Reporting params: {"Fan1":{"Speed":2}}[0m
11:22:28.861 -> [0;32mI (762316) esp_rmaker_param: Received params: {"Fan1":{"Speed":4}}[0m
11:22:28.908 -> [0;32mI (762316) app_main: Received write request via : Local[0m
11:22:28.908 -> [0;32mI (762326) app_main: Received value = 4 for Fan1 - Speed[0m
11:22:28.908 -> [0;32mI (762326) esp_rmaker_param: Reporting params: {"Fan1":{"Speed":4}}[0m
11:22:29.095 -> [0;31mE (762526) esp_local_ctrl_handler: Unable to unpack payloequest handler for esp_local_ctrotocomm_httpd: Data handler failuri: uri handler execution failed[0m
11:22:30.362 -> [0;31mE (763776) esp_local_ctrl_handler: Unable to unpack payload data[0m
```

nravin5, 2023-05-14:
code:.........

 /* Create a device and add the relevant parameters to it */
    fan_device = esp_rmaker_fan_device_create("Fan1", NULL, DEFAULT_POWER);
    esp_rmaker_device_add_cb(fan_device, write_cb, NULL);
    esp_rmaker_device_add_param(fan_device, esp_rmaker_speed_param_create("speed1", DEFAULT_SPEED));
    esp_rmaker_node_add_device(node, fan_device);

    fan_device1 = esp_rmaker_fan_device_create("Fan2", NULL, DEFAULT_POWER);
    esp_rmaker_device_add_cb(fan_device1, write_cb, NULL);
    esp_rmaker_device_add_param(fan_device1, esp_rmaker_speed_param_create("speed2", DEFAULT_SPEED));
    esp_rmaker_node_add_device(node, fan_device1);

................................................................................

19:37:57.619 -> [0;32mI (793736) esp_rmaker_param: Received params: {"Fan1":{"speed1":1}}[0m
19:37:57.619 -> [0;32mI (793736) app_main: Received write reque Received params: {"Fan1":{"speed1":1}}[0m
19:37:58.651 -> [0;32mI (794766) app_main: Received write request via : Local[0m
19:38:14.520 -> [0;32mI (810626) esp_rmaker_param: Received params: {"Fan1":{"speed1":2}}[0m
19:38:14.520 -> [0;32mI (810626) app_main: Received write request via : Cloud[0m
19:38:15.650 -> [0;32mI (811756) esp_rmaker_param: Received params: {"Fan1":{"speed1":2}}[0m
19:38:15.650 -> [0;32mI (811756) app_main: Received write request via : Cloud[0m
19:38:18.512 -> [0;32mI (814616) esp_rmaker_param: Received params: {"Fan1":{"speed1":3}}[0m
19:38:18.512 -> [0;32mI (814626) app_main: Received write request via : Cloud[0m
19:38:25.786 -> [0;32mI (821886) esp_rmaker_param: Received params: {"Fan1":{"speed2":1}}[0m
19:38:27.193 -> [0;32mI (823326) esp_rmaker_par2":1}}[0m
19:38:32.540 -> [0;32mI (828656) esp_rmaker_param: Received params: {"Fan1":{"speed2":3}}[0m

shahpiyushv, 2023-05-15:
@nravin5 , is this with Android app or iOS?

nravin5, 2023-05-15:
> @nravin5 , is this with Android app or iOS?

Android App

nravin5, 2023-05-15:
> @nravin5 , is this with Android app or iOS?

when we open app what ever the first switch we press....like fan or switch ........... then we chance speed the speed param attached with  that pressed switch ............ again close app and reopen app then press some other switch and then change fan param then again param attached with new pressed switch.

sanketwadekar, 2023-05-15:
@nravin5  Thanks for reporting this issue. We are working on fixing this issue.

sebamsc, 2023-05-28:
Similar situation in here, when updating the parameter from the app (Android) the node receives the parameter for a different device.
Reporting with rainmaker.py client is working fine.

khushbushah2302, 2023-05-29:
Hi @sebamsc , we are working on fixing this issue. Will be fixed and released soon.

khushbushah2302, 2023-06-26:
Hi @nravin5 @sebamsc , fix is available in latest release (version 2.11.2).

closed 2023-06-26 by shahpiyushv