espressif/esp-rainmaker #266: Not able to disable MQTT Budgeting on Arduino IDE

PrateetTillu, 2023-05-18:
### Answers checklist.

- [X] I have read the [Rainmaker documentation](https://rainmaker.espressif.com/docs/get-started.html) and the issue is not addressed there.
- [X] I have updated my IDF branch (release/vX.Y) to the latest version and checked that the issue is present there. This is not applicable if you are using Rainmaker with Arduino.
- [X] I have searched the [Rainmaker forum](https://www.esp32.com/viewforum.php?f=41) and issue tracker for a similar issue and not found a similar issue.

### General issue report

I am making an air purifier which updates data of AQI on the app every 3 seconds along with other data of temperature, humidity, etc. This consumes the budget quota quickly. I have tried editing the config file but I am not able to disable the Budgeting. Also what exactly I have to do to disable that. I might be doing something wrong.

shahpiyushv, 2023-05-19:
Frankly speaking, we have observed that mostly its only the Arduino based apps that send too much data to the RainMaker cloud backend unnecessarily which prompted us to add the budgeting and also a [fair usage policy](https://www.esp32.com/viewtopic.php?f=41&t=32450&sid=3f81a59bfe5ecc296a2344459135da34) wherein we deactivate nodes sending more than 25,00 messages in a day.

Even in your case, considering real life scenarios, it seems very unlikely that the ambient temperature, humidity and AQI changes every 3 seconds requiring you to report so frequently. Reporting 3 params every 30 seconds (which is still a high rate) should be sufficient, right?

In case you require a faster update only when the app is open, you can normally report at some reduced rate (say once every 1-5 minutes) and then increase the rate temporarily by setting some boolean param from your app.

shahpiyushv, 2023-05-19:
> 1. Now since I use Arduino IDE I want to know how to use local control. I played with sdkconfig but not able to get it.

Arduino pre-built libraries do not have local control enabled as of now since we were facing issues with it. The team is working on it. You may try the esp-idf based RainMaker in case you are looking for more features and flexibility, since that would have everything in source code.

> 2. Is there any web app to view the parameters too?

No, the [RainMaker dashboard](https://dashboard.rainmaker.espressif.com/) is meant  only for device management and OTA firmware upgrades.

> 3. There is some fluctuations in iOS app toggles but not in android app also no effect on the board itself just the iOS App.

If you have some screen recording of the issue, please share it under [esp-rainmaker-ios issues](https://github.com/espressif/esp-rainmaker-ios/issues).

> 4. Is there a way to send a notification in the app for a situation like too much AQI or Harmful Gas etc.

We have an API [esp_rmaker_param_update_and_notify()](https://docs.espressif.com/projects/esp-rainmaker/en/latest/c-api-reference/rainmaker_core.html#_CPPv434esp_rmaker_param_update_and_notifyPK18esp_rmaker_param_t22esp_rmaker_param_val_t). Even if there is no abstraction for that available in Arduino, you can use this API directly in your code.

> And lastly If we plan for a commercial usage, like customised app, parameters, more nodes for an account, etc, what is the procedure?

Please send an email to sales@espressif.com for the commercial details.

closed 2023-07-12 by shahpiyushv